slic3r/Slic3r #3498: Feature request: Custom, per-filament g-code

VanessaE, 2016-09-07:
I think it would be beneficial to be able to issue a bit of custom G-code on a per-filament basis. For example, if you use firmware linear pressure advance (`M905 K<n>` in Marlin, as of RC7) or retract (`M207`/`M208`), or no doubt other firmware features, some types of filaments may perform better if you could tune these features accordingly.  [The documentation](http://www.marlinfw.org/docs/features/lin_advance.html) for the pressure advance feature, for example, advises using higher K values for flexible filaments than one would use with plain PLA.

I guess any such per-filament code should be inserted into the final file right after the user's custom "start" code.

nicksears, 2016-09-13:
I agree, but I vote for something more universal. We should have script tabs for each section. It doesn't make sense to have all the scripts in the "print" section like we currently do, when you may want to only use a certain script with a certain filament or printer. 

For the printer section it makes sense to have startup/tool-change/end scripts, etc, but not so much in filament and definitely not in print. However, I think we should have a modular section in printer settings where you can have unlimited sections. I posted an example before but I cant find it-we should have a (+) button to add a section that is before/after each print/layer/n layers/retractions etc. I'll remake itto illustrate when I get a chance.

VanessaE, 2016-09-13:
I basically had in mind a text box in the "Filament" section of the profile for a particular filament, to just enter whatever generic g-code one might want to use for that material.

lordofhyphens, 2016-09-13:
Actually a duplication/variant of #3309

lordofhyphens, 2016-09-13:
Also @nicksears it was in #2675.

You can add "involves:nicksears" to find all issues that you've posted in in the tracker btw.

closed 2016-09-13 by lordofhyphens