slic3r/Slic3r #3499: "complete individual objects" freezes slic3r

Tinchus2009, 2016-09-08:
### Version
1.3 DEV (latest) but `git describe --tag` gives me this: 1.2.9-364-g3700950, is this ok?

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Win 10 64b

### Behavior
* _Describe the problem_

It happens a lot of time to me, not only with the 2 object I uploaded as example. When I try to use this option, slic3r freezes. My guess is that is not handling the situation of not being able to print those object in secuence, and instead of showing a warning, it freezes.

* _Steps needed to reproduce the problem_
I load activation_plate.stl and place it on the back of the ronting plate, then I load base.stl and right there slic3r freezes.

  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No
[Documents.zip](https://github.com/alexrj/Slic3r/files/461527/Documents.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2016-09-20:
Can you disable background slicing and see if it still freezes for you?

Tinchus2009, 2016-09-20:
Yes!!!! Disableing background slicing produces no freezing. So the 
solution would be disable background slicing.

Works for me :)

Tinchus2009, 2016-09-20:
This leads me to a question I coudnt find in documentation: If I disable 
background slicing, once Im ready to slice, is there any way to see the 
simlation as always or I need to export the gcode in order to see the 
sliced simulation?

lordofhyphens, 2016-09-20:
I have a branch for that :)

See #3501

lordofhyphens, 2016-09-20:
The root cause probably has something to do with starting/stopping the worker threads on plating change. Still something that probably needs to be fixed though.

I haven't tried to reproduce your issue locally yet (no time -.-), but I knew that there's all sorts of scheduling 'fun' around the background processing. 

Personally, I think that background processing should default to OFF with my PR in place.

bubnikv, 2016-09-23:
From my point of view (limited development resources), I would fix the
thread starting / stopping once I have all the Slic3r computing core
switched to C++, then I would get rid of the perl threads.

alexrj, 2017-03-17:
Background processing is now disabled by default. I'm closing this one as we are aware of some instability with it, and the porting work is ongoing.

closed 2017-03-17 by alexrj