slic3r/Slic3r #3500: Support not generated

skrutt, 2016-09-10:
### Version
_Version of Slic3r used goes here_
1.2.9

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 7

### Behavior
* _Describe the problem_

So, i have tried all combinations of settings for support but still not gotten a good result, expected would be to have support for all the rings but not in the holes. However this seems to be impossible, in the most successful config i turned on enforce support for the first 34 layers and got support for the rings but also inside some holes and also all around the model. 

Setting the treshold to 25 degrees resulted in support inside all holes bur not any under the actual model resulting in most of it floating free.

* _Steps needed to reproduce the problem_
Load model, slice.

  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_
No, but it would be nice to have a "Generate gcode" button when background processing is turned off. 
This would be nice since changing print settings during a slice is for me pretty unstable, and for me slicing this model takes a few minutes. If i want to change 2 settings, i change one and it will start slicing, i then either change the other one and risk a crash/freeze/lag or wait for it to finish which can take a while. Switching views during slicing is usually what triggers a crash for me. (I'm aware that this is not very testable, but if there is interest i could try to set up a reproducible case and send a new issue)
I think there is a memory leak somewhere since slicing this model from a fresh start uses some 300-600 mb ram, but this starts to rise after reslicing it some times and changing the settings between and during slicing. When slic3r was using over 4,5 gb i closed it and restarted, now it's at 3,3 gb again after some 10(or more?) slices. 

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

I could not upload the zipfile as .zip for some reason, so i changed the extension to png. Sorry for the inconvenience but it should work atleast!

[image]

lordofhyphens, 2016-09-10:
"Generate gcode" is called "export gcode" when background processing is turned off.

skrutt, 2016-09-10:
Well sure, but that's not what i was after. There is little meaning for me to also wait for the export of the gcode if, like now, i'm pretty sure it's not gonna be usable? It would be nice to be able to change some settings, hit generate and then preview the slice before the export. Usually i have it on background processing and that's pretty nice, but for some models/cases it's just not very smooth, so i turn it off. But then i basically have to turn it back on again and change something again to trigger the generation. Or like you said, hit export, but since i don't actually want to export anything yet... It's a matter of preference i guess.

lordofhyphens, 2016-09-10:
@skrutt see #3088. 

I always leave background processing off, it causes more frustration for me than it solves.

lordofhyphens, 2016-09-11:
@skrutt see #3501 and #3071 for your reslicing request. 

Support generation is pretty aggressive in Slic3r; it's been on my list of things to tweak. I think that cylinders confuse the bridge detection.

skrutt, 2016-09-11:
Oh alright. I would guess that free floating shells might be sort of easy(relatively) to detect and therefore be a clear target for support, but then again i'm very sure that i'm not the first to think of that so it's probably more convoluted. I tried it with and without bridge detection but it seemed to not make a difference. 

Also i noticed something about the contact distance, for slopes like this it seems to me that the distance is usually more than the settings at? For example, the outmost ring got no support at all with contact distance at 0.2 mm, even though it has three layers of 0.4, 0.2, 0.2?

Great that you have a branch for "slice now" and thanks for looking into this!

lordofhyphens, 2016-09-11:
One request/problem per issue please ;)

If you check through the tracker it has probably already been brought up.

skrutt, 2016-09-11:
That train has long since passed ;)

But you're right, and this is probably caused by a different mechanic.