slic3r/Slic3r #3503: Solid infill of bottom layer is missing

metalzse, 2016-09-13:
### Version 
Slic3r Version 1.2.9

### Operating system type + version 
Windows 10 Pro, Version 1511

### Behavior
* _Describe the problem_ 
Solid infill of bottom layer is missing.
I configured 3 bottom layers in settings, but only 2 of them are actually produced in gcode.

* _Steps needed to reproduce the problem_ 
Normal slicing with provided config.ini

* _Expected Results_ 
I expect that 3 solid build layers are printed after the support interface layer.

* _Actual Results_ 
The first of 3 solid layers is not produced in gcode resulting in only 2 solid bottom layers.

  * _Screenshots from __*Slic3r*__ preview are preferred_ 

Layer Z = 21.20:
Dark-grey is the last support interface layer; whereas blue and red are model layers.
[image]

Layer Z = 21.25
Dark-grey is the support structure, where light-grey is the preview of the next layer.
[image]

Layer Z = 21.45
The problem appears in this layer. As you can see, in the light-grey area, the solid infill surface is completely missing – i.e. the first bottom layer.
[image]

Layer Z = 21.55 
Dark-grey is the support structure, where light-grey is the preview of the next layer.
[image]

Layer Z = 21.7
In this layer, the next solid infill is produced – no problems observed. However, because the underlying solid layer was not printed before (see Z = 21.45), this layer exhibits crappy surface quality.
[image]

The following layers are printed without problems.

#### STL/Config (.ZIP) where problem occurs
[Slic3r Bug Report.zip](https://github.com/alexrj/Slic3r/files/469805/Slic3r.Bug.Report.zip)

bubnikv, 2016-11-28:
This seems to be working fine in Slic3r 1.31.6 Prusa Edition.

The problem in the alexrj slic3r may be caused by the way the Clipper library offset is applied over multiple polygons at once, which is deemed unsafe. The offset shall be applied per contour, then the contour shall be merged.

alexrj, 2016-12-17:
Thank you @metalzse, I fixed this.
It was caused by support_material_contact_distance == 0 and other rare circumstances related to bridge detection.

closed 2016-12-17 by alexrj