slic3r/Slic3r #3505: Mac version doesn't change bed temperature after the first layer

thonghh, 2016-09-16:
### Version
1.2.9 Mac

### Operating system type + version
Mac OSX 10.10.5

### Behavior
* The problem is that I set the bed temperature to 40 degree for the first layer, 0 degree to the other layers in Filament Settings -> Filaments. But the printer doesn't perform what I expected. The bed temp is kept at 40 during the whole printing process.
* Set different bed temps in Filament Settings -> Filaments, but it doesn't change the temp at all

[image]

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2016-09-20:
Could you post a zipped copy of output gcode where this problem occurs? Use something simple like a 10x10 solid box.

phizz166, 2016-09-21:
These values might be getting overwritten by your start gcode. Check that?

thonghh, 2016-09-22:
I just sliced this gcode file, and the problem still occurs. Will check if setting to 10 degree after the first layer solves the problem.

[printer.zip](https://github.com/alexrj/Slic3r/files/486391/printer.zip)

Pointedstick, 2016-10-12:
I'm seeing the same issue on OS X 12.10 with Slic3r 1.2.9, irrespective of whether or not I have any commands to set temperatures in my start.gcode file. The bed starts out heated at the first layer bed temperature, but does not change the temperature after the first layer. 100% reproducible for me.

Pointedstick, 2016-10-13:
Update: this feature works properly when changing the temperature of subsequent layers to anything other than 0. But when you tell it to set the temperature of subsequent layers to zero, it ignores that.

lordofhyphens, 2016-10-13:
Probably because 0 is interpreted as no value. Frankly anything less than
ambient  temp is effectively 0

Pointedstick, 2016-10-13:
Right, I can set it to 10c and that works. But it's kinda weird and unintuitive.

alexrj, 2016-12-16:
The tooltip for those options says: *Set this to zero to disable bed temperature control commands in the output.*.

alexrj, 2016-12-16:
So, we have a dedicated "Has heatbed" option now which determines whether bed temperature commands are written.
However that new option is enabled by default, so if we change the semantics of first_layer_bed_temperature and bed_temperature to consider 0 as a valid value, all current users will have `M190 S0` and `M140 S0` in their output which is bad.
I just relaxed it only for bed_temperature. While it's not totally clean, I think it's a reasonable compromise.

Any thoughts?

Pointedstick, 2016-12-16:
That sounds reasonable to me.