slic3r/Slic3r #3506: Feature Request: variables for Printer/Filament/Printer Settings profile names in output filename

rohieb, 2016-09-17:
### Version
1.2.9

### Behaviour
Slic3r currently allows all config options to be used as template variables for the `output_filename_format` option, as well as additional variables like `day`, `month`, `timestamp`, and `version`, so it would be great if there were additional variables to insert the name of the profiles which were used for slicing.

I suggest the following names:
* `[print_profile]` for the name of the currently used Print settings
* `[filament_profile]` for the name of the currently used Filament settings
* `[printer_profile]` for the name of the currently used Printer settings

For example, I work with multiple printers, so I have at least two printer profiles in Slic3r (named "RepRap" and "LulzBot"), and also several print profiles (like "Coarse with brim", "Fine honeycomb", "Quick print", etc.) So, if `output_filename_format` is set to `"[input_filename_base].[printer_profile].[print_profile].gcode"`, it would generate file names like `myobject.LulzBot.Quick print.gcode`.

lordofhyphens, 2016-09-17:
Was pretty sure it already does.

lordofhyphens, 2016-09-17:
https://github.com/alexrj/Slic3r/wiki/FAQ#how-can-i-specify-a-custom-filename-format-for-output-g-code-files

Yep, there it is.

closed 2016-09-17 by lordofhyphens

rohieb, 2016-09-17:
Ohh, thanks. The popup in the GUI for that settings didn't mention `[print_preset]`, `[filament_preset]` or `[printer_preset]`.

lordofhyphens, 2016-09-17:
https://github.com/alexrj/Slic3r/wiki/Parameter-names-for-Custom-GCode I've called out the ones I know about and stuck them in the wiki.

rohieb, 2016-09-17:
Thanks, but they don't seem to work with 1.2.9 yet? At least I'm getting file names like `_40x10.[print_preset].[filament_preset].gcode` when I export GCode. Is this a new feature?

lordofhyphens, 2016-09-17:
No, it is a fairly old one actually. Are you slicing from the Slic3r GUI or
from something like RepetierHost?

It can't work from RepetierHost.

lordofhyphens, 2016-09-17:
I just verified that print_preset works.

rohieb, 2016-09-17:
Yes, I'm using the Slic3r GUI. After I wiped my ~/.Slic3r/ and re-imported my settings, it now works for me too.

rohieb, 2016-09-17:
Wait no, now I got a literal `[print_preset]` in my filename again… I'm currently trying to find out what it depends on.

rohieb, 2016-09-17:
It breaks when I disable background processing. I will open a new bug report.

lordofhyphens, 2016-09-17:
Try 1.3.0-dev. I leave background processing off.

rohieb, 2016-09-17:
No, still happens for me on current HEAD. See #3507.