slic3r/Slic3r #3507: output_filename_format template variables are not replaced when background processing is disabled

rohieb, 2016-09-17:
### Version
1.2.9-364-g3700950

### Operating system type + version
Debian testing/unstable, Perl 5.22.2, Slic3r from git.

### Behavior
As described in [https://github.com/alexrj/Slic3r/wiki/FAQ#how-can-i-specify-a-custom-filename-format-for-output-g-code-files], when using the GUI, the `output_filename_format` setting can include the print preset, filament preset, and printer preset. However, these are only replaced if background processing is enabled, otherwise the filename will contain the literal variables, like `myobject.[print_preset]_[filament_preset]_[printer_preset].gcode`.

Steps to reproduce:
1. Go to "Print Settings" → "Output options", and set "Output filename format" to `[input_filename_base].[print_preset]_[filament_preset]_[printer_preset].[scale].gcode`. Save the preset.
2. Have background processing enabled, or enable it via "File" → "Preferences", and restart Slic3r.
3. Load any STL, e.g. `GEB.stl`, click "Export G-Code".
4. The pre-defined filename in the save dialog looks like `GEB.Lulzbot fine_Lulzbot PLA DAS FILAMENT_Lulzbot.100%.gcode` (which includes the names of my presets). Close the dialog without saving.
5. Disable background processing via "File" → "Preferences", and restart Slic3r.
6. Load an STL, export G-Code.
7. The pre-defined filename in the save dialog now looks like `GEB.[print_preset]_[filament_preset]_[printer_preset].100%.gcode`, but it should look like before in step 4.

This happens with any STL that I tried (about ~6 different), and also after cleaning my Slic3r settings and doing the Configuration Assistant again to get a fresh preset.

lordofhyphens, 2016-09-17:
It only triggers when you have turned background processing on and then off. If you had background processing turned off via preferences before setting the value it works. -.-

lordofhyphens, 2016-09-17:
Workaround: With background processing off, copy the string for the gcode output, and replace it with something else. Save profile.

Export gcode once with that name, and then go back and paste the old string back and save the profile again. You can now export gcode and it'll give the expected value.

alexrj, 2017-03-05:
Thank you, I fixed this one.

closed 2017-03-05 by alexrj

rohieb, 2017-03-05:
Whee, thanks :-)