slic3r/Slic3r #3508: Unable to access Haproxy-secured OctoPrint

AbuMaia, 2016-09-18:
### Version

Slic3r 1.2.9

### Operating system type + version

Gnome Ubuntu Linux 4.16

### Behavior

Feature Request

My OctoPrint installation's web access is secured as suggested by Haproxy. A username and password are required to access the OctoPrint page. Please add a way to input the credentials into Slic3r so that it may connect and upload to OctoPrint.

lordofhyphens, 2016-09-20:
When you're connecting normally, do you use something like https://username:pwd@<proxyurl> ? I've never set up haproxy (I don't use octoprint myself). I'm also not really a fan of trying to deal with password hashing and saving it in a secure manner (if it's treated like any other config it'd be stored in plaintext in the slic3r config).

Something else you could consider is configuring your proxy to permit direct access to the API url (which needs the API key to work).

AbuMaia, 2016-09-20:
I connect to my IP address followed by :port to use my router's port forwarding to connect to the OctoPi, and haproxy pops up a dialog box asking for name and password. Neither are actually hashed as far as I know, the password was entered into the haproxy config plaintext.

lordofhyphens, 2016-09-20:
Could you try the user:pwd@host:port ?

>
> I connect to my IP address followed by :port to use my router's port
forwarding to connect to the OctoPi, and haproxy pops up a dialog box
asking for name and password. Neither are actually hashed as far as I know,
the password was entered into the haproxy config plaintext.

AbuMaia, 2016-09-20:
Thanks, that works. :) Issue can be closed.

closed 2016-09-20 by AbuMaia

lordofhyphens, 2016-09-20:
Just so you know, the pwd is stored with the config this way in plaintext. Recommend using a different password than you usually use and don't post this part of the config.

AbuMaia, 2016-09-20:
I'm aware. I said as much earlier.