slic3r/Slic3r #3509: Thin details are printed detached when "Detect thin walls" is enabled.

kefir-, 2016-09-18:
### Version
1.2.9
### Operating system type + version
Ubuntu 16.04

### Behavior
I have designed a part with two thin single-extrusion "arms" sticking out:

[image]

When I slice this with "Detect thin walls" enabled, the arms are detached, there is no continuous extrusion from the main body to the arms. Here's a sample layer, all the layers are the same:

[image]

Slic3r should try to prioritize thin sensitive sections of the part like this, otherwise parts will be weak and fall apart. Mine did.

I've reported something similar a long time ago, issue #1237, but I'm not sure this is handled by the same code. The problem is similar, though.

#### STL/Config (.ZIP) where problem occurs

[issue-details.zip](https://github.com/alexrj/Slic3r/files/478884/issue-details.zip)

kefir-, 2016-09-18:
Here's the printed part, by the way. One of the "arms" was almost completely detached. 
[image]

lordofhyphens, 2016-09-20:
Those tabs are 0.5mm wide, and your config's external perimeter width is 0.35mm. With detect thin walls on, Slic3r (correctly) realizes that it can't generate a loop there so it makes a wider trace of 0.5mm.

With detect thin walls off, Slic3r overlaps two 0.35mm perimeters to get it to fit and maintain a loop (which is what it wants). 

In your posted test case, there's no room in the model for Slic3r to loop around back, which is why it's getting sliced as a single thin wall. Given your configuration, the minimum unsupported feature width is 0.7mm (two perimeters). For comparison, Cura uses the same behavior as Slic3r with thin wall detection set to off.

My suggestion is to either pick a smaller printing nozzle or modify the part to make those tabs thick enough to lay down a proper loop. Then, Slic3r will be able lay down a stronger piece.

There is an older branch, thinwall-bpv (referenced in the other github issue), but it's so old it needs to have master merged onto it.

Patola, 2016-09-20:
Well, actually, in the slicing you can see that the path finding algorithm makes the thin wall attached to the greater structure, because its actual spread (the grey area) is continuous within the plotted surface. But as the path itself is not continuous from the big area to the thing wall, in practice this creates a structural weakening. Even if the path algorithm does not turn around when plotting this thing wall, I think it would be effective if it continued from the perimeters of the bigger area to the wall, then discontinued (stop extruding) the path at the end of the wall line, then returned to extrude at the perimeters of the big area again, then continue its path to the end of the second thin wall. Like that:

[image]

lordofhyphens, 2016-09-20:
Give it a try with the old branch referenced in the other issue.

kefir-, 2016-09-20:
So if I make the tabs/arms 0.35mm, and set my external perimeter to 0.35mm, it should make it contiguous? I see the logic in that. But I also think it'd be an improvement if slic3r could make the outer perimeter along the main part 0.5mm where necessary, in order to get these attached,  provided we're not losing any detail on the outer perimeter, which I don't think we would in this case. I don't recall right now if I had set a specific outer perimeter width, I'll go back and try some more.

lordofhyphens, 2016-09-20:
You need to make the tabs/arms 0.7mm with an external perimeters of 0.35mm to get a contiguous loop around the whole thing. I looked at the config, you have pretty much everything fixed in width. 

I think that that smart perimeter thing in the  thinwall-bpv branch is a solution to your problem, especially if the recent changes to permit variable width perimeters are also incorporated.

bubnikv, 2016-09-23:
I completely agree with @patola, that Slic3r shall somehow re-connect the
lines as on his picture. This requires an additional step of building up a
graph of perimeters with the thin fills attached and traversing this graph
by some heuristics. Not a simple task though.

alexrj, 2016-11-24:
Did you test this with 1.3.0-dev?  It might be improved than 1.2.9 as I did much work on it.

lordofhyphens, 2016-11-24:
@alexrj I replicated the problem myself back in September with 1.3.0-dev.