slic3r/Slic3r #3511: Feature Request: More support patterns

lordofhyphens, 2016-09-20:
As it turns out, Support material patterns use the Fill patterns. 

[image]
[image]
Is there any interest in these more exotic patterns? 

It's not ready for a PR, but https://github.com/lordofhyphens/Slic3r/tree/all-infill-support has my small hack to add the support materials to the drop-down.

I think for people desiring less sturdy support fills, removing the external loop generated may be sufficient to make easily deconstructed support.

Hope someone finds this useful to experiment with.

Patola, 2016-10-07:
I have a big interest in this. The lines support pattern with no interface makes the support behave quite like Simplify3D's support, which is very easy to remove.

kefir-, 2016-10-07:
What @Patola said! I've used the lines pattern (I think, lots of parallel lines simply connected at the end, every layer in the same direction), and when I managed to get slic3r to create what I was looking for, I was very pleased with the result.

limawill, 2016-10-07:
I have a big interest in this!!!

bubnikv, 2016-10-07:
This is what I am currently working on, I have an early prototype, which
needs to be polished.

Joberr, 2016-10-11:
That's a very good idea!

bubnikv, 2016-11-21:
The Prusa3D build has a feature "without sheath" and a new zig-zag infill pattern. With that, you can configure the supports very much the same way as Kisslicer or Simplify3d behaves.

Patola, 2016-12-23:
To be more clear in what I said, to me the ideal support pattern would be the rectilinear2 without sheath - which is exactly like simplify3d's pattern.

Sensemaya, 2017-03-07:
Pillar patters need to have a wider base support in the first 3 layers then goes thin, I notice some pillars will fall when extruder deposit more material on top

NateTG, 2017-04-01:
I imagine that a space frame like the 'wire frame print' would be a pretty good support material.  It does have the drawback that you can't print it close to the rest of the part.

Kolodner, 2017-12-22:
I'd just like to be able to dial in things like the spacing between pillars and the pillar shapes... I've had some nightmare pillars that were impossible to remove that bonded too firmly to their neighbors. If we could specify a minimum gap to leave between pillars, that'd be great.