slic3r/Slic3r #3512: Spiral mode not working

Tinchus2009, 2016-09-21:
### Version
1.3.0 dev latest (main branch)

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
win 10 64b
### Behavior
. Spital mode seems to be broken, I have tried several files and basically the same result. I have atached the latest test object. Perimeter are set to 1 (of couse), detect thin wall has been deactivated the same as add extra perimeters. And of course spiral mode is activated. 
The problem starts at the sliced simulation: you can see 2 perimeters being created, first warning sign. When printing starts, the printing starts like a normal print: 2 perimeters are printed, no spiral movement is detected. After like 20 layers, printing switchs to spiral mode but still 2 perimters are printed, and after 20 layers or so, again printing switchs to normal print, and this switching process repeats till the end.

[Left_Claw_1.zip](https://github.com/alexrj/Slic3r/files/485030/Left_Claw_1.zip)
[config_spiral.zip](https://github.com/alexrj/Slic3r/files/485032/config_spiral.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2016-09-23:
Its a funny coincidence. I fixed a bug in the spiral mode today.
https://github.com/prusa3d/Slic3r/commit/565146d9d1be222693901a26acb5a3c65dd8ac65
Is it the same issue?
Vojtech

Tinchus2009, 2016-09-23:
Dont know if it is the same, your fix has been added to the main branch?

bubnikv, 2016-09-23:
> c65dd8ac65

To the main branch of what? I don't have access to the alexrj github, my
bugfix is at the main branch of github/prusa3d/slic3r.
The change is tiny. You can easily copy it to your code.

Tinchus2009, 2016-09-23:
Well... Im not skilled in useing git or in code. I just did a git pull 
command and your fix is not there. I Prefer to stay on the oficial side 
of the fixes and modifications. Your fix should be added soon I guess

bubnikv, 2016-09-23:
I would be thankful, if you could test the fix.
Just add that single word into lib/Slic3r/GCode/Reader.pm , no github
knowledge required.
You will find the line to change looking at the following page:
https://github.com/prusa3d/Slic3r/commit/565146d9d1be222693901a26acb5a3c65dd8ac65

Vojtech

Tinchus2009, 2016-09-23:
Ok, I will test it: I change that part of the code, and then ? I have to 
rebuild slic3r or I can just change that part of code and launch slic3r?

bubnikv, 2016-09-23:
Either change the line in the source directory and rebuild, or change it in
the installation directory.
The code is not compiled, but it needs to be installed, which is normally
done at the end of Slic3r compilation.

lordofhyphens, 2016-09-23:
I will have a PR for this up when I get some time.

Tinchus2009, 2016-09-23:
It doesnt build now with the modification. I get this error:

t\shells.t ..................... 6/21 Can't locate object method 
"90.300" via package "Slic3r::GCode::Reader" at 
C:/Slic3r/t/../lib/Slic3r/GCode/Reader.pm line 25.

alexrj, 2017-03-20:
@Tinchus2009, this is not a bug. Your model contains a cavity inside, so two loops are required for each layer and spiral vase doesn't apply:

[image]

closed 2017-03-20 by alexrj