slic3r/Slic3r #3515: Slic3r ignoring "slow down if layer print time is below" setting

phizz166, 2016-09-23:
### Version
1.30-dev

### Operating system type + version
Windows 7

### Behavior
I'm printing in an extremely low-temperature (90C) caprolactam filament, which needs extensive cooling to print well. In addition to 100% fan usage, I'm trying to tell Slic3r to slow down the print and maintain a minimum layer time of 30 seconds. However, no matter what is set as the minimum print speed, I get the exact same file every time. Slic3r appears to be ignoring this setting.

I've attached three copies of generated gcode and the STL for a simple 20mm cube. All of the settings in each file are identical except for _min_print_speed_, which is set to 10/5/1. _slowdown_below_layer_time_ is set to 30 for all three. The rest of the generated code is bit-for-bit identical. Printing any of the three gcodes results in a layer time (for the middle, sparse-infill layers) of ~10.5 seconds, so obviously it isn't responding to the 30 second limit. Furthermore, it takes close to 2 seconds for one outer perimeter to print, so with a total length of 20+20+20+20=80mm, that implies that it's printing at the  set perimeter speed of 40mm/s -- meaning it isn't hitting the lower speed limit either.

Not sure what's going on, but it's ruining prints with small perimeters and short layer times in this material. Is there a setting I'm missing somewhere, or is this just a bug?

#### STL/Config (.ZIP) where problem occurs

[gcode.zip](https://github.com/alexrj/Slic3r/files/490790/gcode.zip)

ch100, 2016-09-26:
Have you tried to apply this revert? https://github.com/alexrj/Slic3r/pull/3443
It should work then.

Tinchus2009, 2016-10-03:
Im oserving the same issue here: slic3r is completely ignoring the cooling thresholds settings, the "slow down if layer time is..." option is not being used and the "min print speed" also is then not being aplied.
This is somehow critical, many prints have small layers and in the case of PLA, if there is at least 10 seconds of cooling before another layer is deposit on top, heating issues arise.

Tinchus2009, 2016-10-03:
@ch100 can yo explain how to apply that revert?

ch100, 2016-10-06:
I don't know which platform you're using, but there's no magic. Apply the patch and rebuild Slic3r. I really hope that the revert works for that problem too.
If you don't know how to apply a patch, I would suggest you to wait until this bug is being fixed.
Alex and other people are making this great software in their spare time. Making software takes time and patience is a virtue.

JBKingdon, 2016-10-09:
Hi, just a "me too" post. I noticed this yesterday on a testcube while I was checking calibration. Fan didn't increase above minimum setting and speed wasn't reduced, so layers were 4s instead of specified minimum 10s. Build was from the master on Sep 27.

VanessaE, 2016-11-02:
Still a problem at 3700950474583e6c64163fab848a831cdcfbf44e.

alexrj, 2016-11-22:
Fixed, thank you for the report.

closed 2016-11-22 by alexrj