slic3r/Slic3r #3516: Suggestion for a feature

Shaggysgti, 2016-09-26:
Hey you guys, love Slic3r! It's been my jam since I've started 3-4 years ago and I've never needed to defect. I recently had an idea that I think could be implemented for the next update. My idea is to interleave the perimeter walls and the infill. So on layer one, we can have 5 shells with full infill and then the next layer could be 3, or 2 layers, so that the infill will overlap and interleave with layers, quite like brickwork.

lordofhyphens, 2016-09-26:
Could use a picture. Also how do you think it would work with sloping
surfaces?

We do already have overlap in x/y.

The good news is that you can test this plan. Generate 2 copies of the
model with different perimeters settings and verbose gcode on. You can then
interleave the two, layer by layer. That will give you a good idea if it is
worth following up any more.

Wouldn't take more than an hour to cook up a script to automate that part
of the process, I think; straightforward text processing.

>
> Hey you guys, love Slic3r! It's been my jam since I've started 3-4 years
ago and I've never needed to defect. I recently had an idea that I think
could be implemented for the next update. My idea is to interleave the
perimeter walls and the infill. So on layer one, we can have 5 shells with
full infill and then the next layer could be 3, or 2 layers, so that the
infill will overlap and interleave with layers, quite like brickwork.

lordofhyphens, 2016-09-26:
https://github.com/lordofhyphens/gcode-blender is such a script for blending 2 similar pieces together. I hope you find it to be useful.

djdelorie, 2017-04-04:
FYI I've needed this before, and used modifier meshes to set an alternate number of perimeters for every other layer.  Makes for much stronger prints when you can't afford the weight, like quadcopter blade guards.  No need to change slic3r to do it, though.

foreachthing, 2017-05-11:
I'd like that too!
I always use that in Cura (_alternate extra wall_). I like to think, that the part prints faster, too (_this_ part prints 10 minutes faster (total of 2:20 h, instead of 2:30 h))! On one layer 3 perimeters (shells), the next layer only two and the next three again - and so on.