slic3r/Slic3r #3517: Feature request: Make support material attached to the bed have wider and stronger bottom layers

mrjohnc, 2016-09-26:
Hi

This is a request for a feature to improve the adhesion and strength of support material attached to the bed. 

Issue: The only sections of prints I ever have problems with coming away from the bed is the support material, this is because they are tall thin towers with little bed surface area and weak bottom layers. Even with a large brim around the objects this does not solve the problem because the second layer gets a lot of force and breaks off.

Possible solution: Where possible increase the surface area of the support material that is attached to the bed and the strength of the lower layers, possibly by giving it flared bottom layers or increasing the thickness of the lower layers or both. Neither of these suggestions will make it more difficult to remove support material from the models.

Thanks

lordofhyphens, 2016-09-26:
I'm pretty sure this is a duplicate of another request, but not entirely sure atm.

lordofhyphens, 2016-09-27:
* Support layer thickness is already maximized. 
* Flared bottom layers isn't a terrible idea, but you'd need to extend the outer clipping solid. otherwise you're basically back where you started.
* If you're using the pillars support type, there's a PR floating around to change the pillar size/spacing. #3485 
* a cheap way to get a larger base is to scale down to the actual support area, see generate_pillars_shape() in lib/Slic3r/Print/SupportMaterial.pm

Bendimon, 2016-10-01:
@mrjohnc Depending on the height of the surfaces that need support material, I choose different support strategies. I think pillars are fast and cheap in terms of plastic consuming. Optimal, but risky. I often choose that risk, but when I leave my printer working over night with a big part, sometimes I prefer to choose, for example, a rectilinear pattern. In that cases, I prefer to waste a little more time and plastic, and avoid a bad surprise the next morning.

bubnikv, 2016-10-07:
You may try to build the whole print on a raft.

bubnikv, 2016-10-07:
> Support layer thickness is already maximized.

Not exactly. The layers are generated for the whole object globally and the top / bottom surfaces have their height defined by the bridging flow and the support gap, so they are not a multiple of anything. So if you have too many top / bottom surfaces at various heights, they add thin layers to the support, contradicting the effort of making thick layers. The current implementation works to support horizontal bridges, but not so well for supporting sloping surfaces.

soswow, 2016-10-14:
And while we there, maybe we could make support first layer spacing configurable.
I've spent almost two days trying to make first layer of quite small support structure to stick to the bed. I already operate on speed 10mm/s for the first layer, and it still doesn't help. It perfectly prints everything else (solid fill, brim, everything) but lines in this zig-zag pattern are not far apart to make it just a line, and not close enough to make it like solid infill. And my hotend start dragging it around.

[image]

cangelis, 2016-10-14:
Building supports on the raft may be an option that is not supported by Slic3r currently. https://github.com/alexrj/Slic3r/issues/3049

bubnikv, 2016-10-14:
Stay tuned, I am hacking the supports right now.

soswow, 2016-10-14:
@bubnikv I will. https://github.com/alexrj/Slic3r/blob/3700950474583e6c64163fab848a831cdcfbf44e/lib/Slic3r/Print/SupportMaterial.pm#L746 seems like the number I want to change. Tomorrow I'll try to build slic3r locally (all the proper Perl version installation hell) and see if it even helps my situation.

bubnikv, 2016-10-14:
> Slic3r/blob/3700950474583e6c64163fab848a831cdcfbf44e/lib/Slic3r/Print/
> SupportMaterial.pm#L746 seems like the number I want to change. Tomorrow

I am currently rewriting the support code into c++. I was unable to fix the
original perl implementation of supports to do the support on object
interface correctly as it requires more information to keep per layer than
just the polygons and the z coordinate. And I plan to rewrite all the
bacakend into c++ anyway, so it was a good opportunity for the supports to
do it now.

soswow, 2016-10-15:
eh ... after 3 hours trying to install everything (perl, perlbrew, local-lib), when on every step something was failing, so I had to use --force or -f ... it still failed in the end. I quite for now.

7eggert, 2016-11-21:
This seems close enough so that I'll add my observation:

1) Slicer seems to actively avoid connecting the support to the brim, and thus it breaks away. It does the same for the actual piece I print, but with an actual air gap that does not  hold small pieces in place. I think one thread to the part is needed at least. I'm using raft for now.

I attached a picture with brim and support showing the problematic parts.

1b) The pillars stick to the parts quite well ... at the places where there should be a gap.

2) The brim itself does break apart. I'd like to add a second layer, e.g. the 20 % linear infill. above the brim.

3) Also I'm currently printing my third piece (ABS was a mistake for beginners). It has minimal bottom area, and I am wondering: Why not just fill the whole bottom layer except where the part is touching the surface? First layer completing the circles from the brim, second layer raft/infill (off cause with gaps for the parts if they are starting at layer 1).

4) For high pillars, I imagine combining them to one large foot might be a good thing.

[image]

lordofhyphens, 2016-11-21:
The brim is just on the outermost contours of the object.

Could you add an example model and settings where you see this problem?

7eggert, 2016-11-21:
http://7eggert.selfhost.bz/tmp/slic3r/brim%20breaks%20apart/

Sorry for the potato-cam.

lordofhyphens, 2016-11-21:
Potatoes gotta potato. :)

 BTW, you can drop images and ZIP archives here. I've edited your post to include them inline incase your host goes down for some reason (which has happened with other things).

7eggert, 2016-11-21:
OK, wasn't sure weather the size would be permittable here.
Also: The layer height is 0.3 mm.

7eggert, 2016-11-21:
More information: It might be due to bad Z calibration due to heating up:

http://shop.prusa3d.com/forum/prusa-i3-kit-building-calibrating-first-print-main-f6/live-z-adjust-constantly-decreasing-t2118.html

Being the proud owner of the probably-best-calibrated heatbed is no lasting feeling ... but at least I know what's happening.

TLDR: The current brim should be enough, but real life happens and a reinforced base might safe some prints.

lordofhyphens, 2016-11-22:
Most things compress down pretty well; the layer height should be captured in your configs as well.

7eggert, 2016-11-22:
I now confirmed that by leveling the _hot_ heatbed twice using the script below and then printing something (not that high, but it looks much better). Unfortunately during the second print, there was a sudden change of z axis, while printing the about third layer, the threads started hanging in the air.

https://github.com/7eggert/3d_printer_leveling_ABS

(Yes, the layer height etc. is in the config, but I'd not like to download everything just to see that.)

7eggert, 2016-11-24:
A few prints later I can add that layer height > 70 % nozzle width will cause delamination. With 2/3 - 0.2 mm having a 0.3 mm nozzle - I get good results (only small prints in the meantime) and a brim sticking to it's neighboring objects.

OTOH, ABS does shrink and pull itself and the brim from the surface. Especially using my test file (edited to 2 cm height instead of changing the model), I get misprints. It's a one-thread 10×10 square with brim - or something curly. I don't know weather a strong clover leaf brim might help, but printing coins at the edges is suggested. I tried printing at 230 °C too, but then the layers don't connect that well.

Since there is the shrinking issue, I now think that connecting the supports may cause the support to be pulled away.

[rect.gcode.gz](https://github.com/alexrj/Slic3r/files/611146/rect.gcode.gz)

bubnikv, 2016-11-28:
> Another option would be to have bridges at different layer heights between nearby support material towers

https://members.loria.fr/JDumas/files/papers/scaffoldings/bridging-the-gap.pdf

VanessaE, 2017-11-25:
Issue is over a year old now.  Can we *at least* have an option for 100% solid bottom layers on supports?

martin-git, 2018-07-18:
Agreed, just a solid bottom would be great! Or an override of the bottom interface layer where the brim should go. Wherever there is support close to the base of the part the brim is replaced by the interface layer. I am having loads of troubles printing PC which tends to warp a lot. No chance without a rock solid first layer+brim.

MortarArt, 2018-08-31:
This really needs to be done. I've reported it elsewhere, and it's the single biggest obstacle to using slic3r, for the huge variety of prints I do as a service bureau. I am very often hacking this, producing custom brims, splitting gcode files, etc. to achieve this, because it is the only way to print a large number of materials and geometries.

supermerill, 2019-01-19:
see https://github.com/slic3r/Slic3r/issues/3256

pileofrogs, 2019-04-26:
++ It's 2019 and this still breaks my prints

HeyRobb, 2020-02-15:
Same here... please add this feature.