slic3r/Slic3r #3520: [Feature Request] Create vertical sections with different profiles

Bendimon, 2016-09-30:
On Plater-->Settings, I see I can create sections overriding default layer heights.
For me, it would be more interesting to be able to choose the profile used in each Z interval.
At least, to choose a 'print settings' profile. It would let me to vary infill, print speed, and that 'layer height'.

I know modifier meshes do the job, but I find too many situations (almost all) where my wish of change parameters is vertically. In theese cases, my request would be easier than using modifier meshes.

unmarcos, 2016-09-30:
Yes! Very interesting request.

lordofhyphens, 2016-09-30:
I have an openjscad script in the util directory that lets you specify the
mesh.

I also know there are other open issues asking for this.

Patola, 2016-09-30:
@Bendimon  Is this because of Simplify3D's processes system, where you can choose the layer heights for which a particular configuration is valid?

I think that for it to fit within slic3r's workflow the interface would have to offer redundant settings. I agree that the modifier meshes are a bit more difficult for your particular case than just specifying a layer interval, but they are a much more powerful and general case and should be used instead.

lordofhyphens, 2016-09-30:
https://github.com/alexrj/Slic3r/blob/master/utils/modifier_helpers/layer_generator.jscad

Drag that file onto the window at openjscad.org.

A box to specify several settings should show up.

Bendimon, 2016-09-30:
@lordofhyphens Thank you for the link, but I requested a feature in the software, not an external sollution. I know how to solve my need outside slic3r, only thought it could be good to solve completely inside slic3r.

I can't figure how difficult it could be to do this. I don't think it would change the actual workflow, since is something similar to what we already have, but with more options to change.
Seeing from a different point of view, it could be as if I could put different parts, with different parameters for each one, but having that parts in a vertical stack.

...sorry for my bad english. If something sounds rude, it will be because of a mistake.

lordofhyphens, 2016-09-30:
It is mostly UI-related, I believe. Slic3r can already generate simple
shapes (it does so for testing purposes).

Implementing the UI to support it is nontrivial, and I have seen some
movement by @alexrj to use a C++ based ui instead of perl/wx perl.

I will look at pull requests, but it is a low priority given that there is
a straightforward workaround and there is enough to do as it is.

Bendimon, 2016-10-01:
@lordofhyphens I have tried your proposal of generating meshes, and I would like to talk about what was my problem:
When I design a part, It's not always oriented as I want to have it on the printer. And its origin is not on his base, as I design severall parts in an assembly. Or perhaps, my part has been drawn by somebody from thingiverse.
I open my part on slic3r, and have to make some transformations (manual rotations, slic3r automatically translates the part to have it centered, and laying over Z0).
Then I import a modifier mesh, and I see It isn't where I wanted.

I'll try to explain my case with an example:
On OpenJSCAD I create this part:

function main() {
   return [
       torus({ ri:25, ro:40}).rotateX (90).translate ([200,0,50])
   ];
}

To slice it,I want to rotate 90º around X.

Now I want to create a modifier mesh that aplies from Z2 to Z7. Using your script, I set layer_height to 5 and Z=2.

When I open this cube as modifier mesh, I see it isn't at Z2, perhaps because this mesh was automatically transformed.

I would like to know what I'm doing wrong.. or If we should be talking about another request isue, concerning an option to decide if modifier mesh has to be transformed or not.

I'm sorry if I'm being so boring..

Tinchus2009, 2016-10-03:
Well, I have not tried that openscad scropt, but you problem is related to the coordinates system of the model and the openscad object you creat, they dont match and that is why once you import the modifier into slic3r it appears in the wrong position. Regarding the comment about simplify3d.... the modifier tool in slic3r is way more powerfull, because it lets yu work on specific areas of the model, and simplify only lets you specify a range of layers.
Returning to the original issue, my way of dealing with coordinates is useing solidworks (I use this but you can use any other cad software that lets you import stl files). I import the stl, then I build the modifier (the advantage I find here is that I can build a modifier looking the stl, so it will be exactly how I want it and located where I wanted). Then I just export each model independtly and they will share the relative coordinates. I use modifiers in 70% of the models I print and never had a problem working like this (apart from the bug I posted related modifiers )

lordofhyphens, 2016-10-03:
If you import/export an stl in slic3r it will put it at the new coordinates
on the plate, fyi. So if you have a 0,0 = center of bed plate that makes
things very easy

lordofhyphens, 2016-10-03:
I agree that being able to generate/place the simple modifiers via the ui
is quite important to a lot of people's workflows.
On Oct 3, 2016 6:47 PM, lordofhyphens@gmail.com wrote:

> If you import/export an stl in slic3r it will put it at the new
> coordinates on the plate, fyi. So if you have a 0,0 = center of bed plate
> that makes things very easy
>

Tinchus2009, 2016-10-04:
I will try tthat. Right now my 0,0 is the left corner of my plate

Bendimon, 2016-10-04:
@Tinchus2009 Your 0,0 is the left corner of your printer. But you've told slic3r the dimensions of your print area. Slic3r will generate the Gcode offsetting the values to the center of the given bed size.
For example: If you slice a 10x10x10 cube, we shoud expect XY values of +5 and -5. But if you inspect the gcode, you'll see the difference.  It will be something like (bed size/2)+5, (bed size/2)-5 for each axis.

Tinchus2009, 2016-10-04:
Yes, my bad, I meant a different thing in the last answer. I understand 
what you mean. I dont have issues with the modifier's location. But it 
is true that if you have and stl of unknown source, if you make a 
modifier useing the script lord hyphens provided us or any other CAD 
soft like openscad for example, when you import the modifier into slic3r 
it will not match the location with the model.

At the beginning I had this problem and some googleing gave me an answer 
like the problem was relative coordinates inside the model and the 
modifier or something like that. The solution was opening the model with 
any cad software, open and postion the modifier and save each one of 
them again, so the cad saved the same coordinates system for both object.

Anyway, modifers works very nice for me.

lordofhyphens, 2016-11-27:
See #3590

Tinchus2009, 2016-11-27:
Looks really great, I have to wait till the build problems under win is solved adn tehn I will test it, but so far looks like this feature has been implemented?

alexrj, 2016-12-18:
@lordofhyphens's code is merged. Creating stacked vertical modifiers is much easier now!

closed 2016-12-18 by alexrj