slic3r/Slic3r #3521: Set bed size /0

Taragorm, 2016-10-02:
### Version
1.2.9

### Operating system type + version
Linux  OpenSuse Tumbleweed x64

### Behavior
* _Describe the problem_
When in the "set bed size" dialog, cursoring to the first character of X and hitting delete causes
an immediate crash with a divide by zero exception. I.e. when 200 -> 00 in the edit box.

* _Steps needed to reproduce the problem_
See above

  * _If this is a command-line slicing issue, include the options used_
N/A

* _Expected Results_
It shouldn't crash.

* _Actual Results_
It does.

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
N/A

bubnikv, 2016-10-02:
I hope this has been fixed by me.You may try the Prusa3D build
https://github.com/prusa3d/Slic3r/releases

or the builds of the official Slic3r made by @LoH
https://bintray.com/lordofhyphens/Slic3r/Slic3r_Branches

Vojtech

alexrj, 2016-11-22:
Yes, AFAIK this was fixed in 1.3.0-dev.

closed 2016-11-22 by alexrj