slic3r/Slic3r #3522: Does Slic3r cope with large build volumes?

Pat-McD, 2016-10-02:
### Version
_Version of Slic3r used goes here_
1.27 Precompiled 
About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 16.04

### Behavior
* _Describe the problem_

I have a question about Slic3r and support for large print volumes. Cura handles these without a problem, but using the 1.27 precompiled release on Ubuntu, Slic3r crashes on loading a large volume STL every time. Meshlab also handles big files without a hiccup. Setting up a big print volume on Slic3r seems pretty easy (no maximum Z input?) but the program just hates big STLs. Tried other formats, same problem (although crash not always straight to desktop). Slic3r just displays a "Processing" window for a few seconds and shuts down without an error report.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Pat-McD, 2016-10-02:
[PatMcDSli3rIssueLargeVolume.zip](https://github.com/alexrj/Slic3r/files/504925/PatMcDSli3rIssueLargeVolume.zip)

Pat-McD, 2016-10-02:
32 bit Ubuntu and Slic3r, I hasten to add. Tried a compile but my Linux skills were not up to the task.

bubnikv, 2016-10-02:
As far as I am know, Slic3r will run out of memory on large volumes, if
compiled for a 32bit architecture. 64bit slic3r shall be fine.

Also there is a problem with the 3D visualization, where Slic3r tries to
allocate huge pieces of linear memory for vertex buffers, which fails. I
have fixed this in the Prusa3D fork, but I have not made an official build
yet.

Vojtech

Pat-McD, 2016-10-02:
Thank you. I can replicate the error if I load a good STL of a printer part into Slic3r and try to scale it up for a massive print bed... So I'm pretty sure it's a current Slic3r limitation. The reason I opted for 32 bit is that previous Slic3r releases have not liked 64 bit. I'll give it a go.

lordofhyphens, 2016-10-02:
I have been looking at porting the vertex buffer changes into a pull
request but it is tangled slightly with other changes in the code. That and
I have been buried in work and research.

Pat-McD, 2016-10-02:
Yes, I figured it was on the "to do" list. It's just that big build plates are now becoming available... Keep up the good work, Slic3r is still miles better than when I last ran it 18 months ago. :)

Pat-McD, 2016-10-02:
I can replicate the problem by taking a small STL previously saved by Slic3r and trying to scale it up. So it's more of a limitation of the program and large volumes than a bad STL.

BuilderGuy1, 2016-10-06:
I've encountered this problem from a different angle that I thought I should mention. I'm using a Peachy Printer (laser based top-down SLA) and I set the Layer Height and Nozzle diameter to 0.02 mm. This locks the cpu at 50% on my Mac book Pro (10.11.6) for as long as I let it run (12 minutes then killed it). This creates a huge number of layers to manage just like a large build area.

lordofhyphens, 2016-10-06:
I wasn't aware that there were any Peachys in the wild.

Thanks for the heads up though. :)

BuilderGuy1, 2016-10-14:
There are currently only a small number out in the wild but there were ~5000 reserved in the Kickstarter. Once that gets resolved and the orders filled, there will be A LOT of people looking for a good slicer ;-)
You can check out my misadventures with the Peachy here:
http://3dprintboard.com/showthread.php?24028-Delivery-No-1-Build-blog-of-the-first-V1-0-Peachy-Printer-in-the-Wild
By the way, I REALLY like Slic3r and hope to use it once this issue is resolved !

bubnikv, 2017-01-27:
First, the Slic3r has many "epsilons" built in, which are tuned to a usual FDM nozzle diameter. If you set the nozzle diameter too low, chances are a lot of the data resolution will be below the "epsilons".

Second, there seems to be a problem with the Clipper library and the offset function. If called over huge polygons, it eventually takes forever to finish a single operation. This is what kills the current Slic3r supports. I tried to run your test on the latest Prusa3D 64bit Slic3r build and it hangs on infills forever. Probably just doing the offsets per polygon, not over all polygons at once, would improve the speed significantly. I would have to do some profiling to confirm my hypothesis.