slic3r/Slic3r #3523: Incorrect internal structure from some STL files.

vvranjek, 2016-10-03:
For all versions including 1.2.9

It doesn't show the correct internal structure of the attached STL files. To reproduce the problem just open the files in Slic3r and in some other program and see the difference in the internal structure.

Let me know if there is anything I can do help.

Cheers

[Slic3r bug report.zip](https://github.com/alexrj/Slic3r/files/507186/Slic3r.bug.report.zip)

Patola, 2016-10-03:
These objects are full of mesh errors and aren't even closed meshes... And full of absurdities like shifted thousands of units in one axis... fuselage 1 conversion: 464 non-manifold edges, 2642 intersecting faces... wingL_1: 464 non-manifold edges, 1850 intersecting faces... Blender even does a decent job of displaying it and filling it up internally, in the case of fuselage 1 the outer "wall" is not a closed mesh around the shape, it's just a surface with no bottom.

When you complain about the internal structure are you referring to the sliced result (Preview tab) or main window (3D)?

vvranjek, 2016-10-04:
> When you complain about the internal structure are you referring to the sliced result (Preview tab) or main window (3D)?

I was referring to 'Preview' and 'Layers" tab since I don't think you can see inside in '3D' view.

> Blender even does a decent job of displaying it and filling it up internally, in the case of fuselage 1 the outer "wall" is not a closed mesh around the shape, it's just a surface with no bottom.

I had no luck displaying the files in Blender at all, but I have  never used blender before, so I had no idea what I was doing. However, it opens just fine in FreeCad 0.16, Cura 2.1.3  and Simplify3D 3.1.0. Also, these STL files aren't supposed to be filled internally (they are hollow) other then the designed  structural components, most of which don't show up in Slic3r.

> These objects are full of mesh errors and aren't even closed meshes... And full of absurdities like shifted thousands of units in one axis... fuselage 1 conversion: 464 non-manifold edges, 2642 intersecting faces... wingL_1: 464 non-manifold edges, 1850 intersecting faces... 

I am no mesh expert, so I'm not sure what these absurdities mean, but I assume that it can't be that bad, since no other program I've tried complains about it. I think Slic3r should be able to do it to.

P.S. 
I didn't create the designs, I bought them for $20, so I assume they are made by professionals.
I am just trying to help you guys  make Slic3r an even better software then it already is.

Patola, 2016-10-04:
Sorry, when I said that blender even does a decent job of displaying it, I actually meant slic3r. You can't see inside the object in the 3D view but if you open the cutting window you can.

Blender (yes, now about Blender) did not display the file because, as I said. it is shifted thousands of units to the side, so it does not appear in the main view - it is even outside clipping distance. So to view it in Blender you have to bring up the Navigation menu by pressing "N" then look at its X-Y-Z coordinates, and change them all to 0.

No, I am not one of the slic3r developers - I am just a guy like you who felt entitled to comment, so I am not voicing any general opinion here. I've taken a close look at the STLs you shared here and frankly, they are terrible, I do not doubt they are made by professionals because there are just too many bad professionals on the market, specially in the 3D printing field - messed-up meshes are due to people who were educated and graduated in making good-looking meshes - appropriate for 3d renderings or simulations - but no good for 3d printing. There is some amount of automated fixes that might be employed (which is what makeprintable, microsoft azure and netfabb do, and most slicers to an extent too), but for your meshes particularly, they will not work, because they are way too messed up, sorry to inform you.

I have analysed them in ccViewer (from cloudcompare), Blender, meshlab and netgen.

bricofoy, 2016-10-28:
Hi 
I also have a file that does not work at all with slic3r, the inside is fully empty, it computes only the walls.
It's not a visuel rendering problem, the produced gcode is also only for the outside walls.
The file was created be me with freecad 0.16, and is opening correctly in cura. As it is my very first try to make a 3D part, it might be very well my design is broken. But it works well in cura...

[support.zip](https://github.com/alexrj/Slic3r/files/558177/support.zip)

bubnikv, 2016-10-28:
Vid,

would Cura produce a correct result for your model? My experience shows
that Cura is better than Slic3r in regard to handling bad object and we
shall learn from Cura.

Vojtech

bricofoy, 2016-10-28:
regarding my file, it prints ok with cura. with slic3r, no matters the settings, it produce only the outer walls, and the part is empty inside

vvranjek, 2016-11-08:
Yes, cura has no problem displaying the file.

alexrj, 2016-11-22:
Files supplied by @vvranjek are highly invalid and they describe a very ambiguous geometry. When ambiguous geometry is supplied, behavior is undefined (= random).  Automatic repair in such cases is a matter of guessing what user expects, among all the possible repairs.

Look at the first file:
[image]

Even my brain is not able to understand what you expect from such file.

The external shell is one-dimensional, with no thickness, and then it contains a structure of internal walls. Do you want it to be entirely filled regardless of the walls? Or do you want an automatic thickness to be applied to the outer shell?  (What thickness? How would Slic3r guess it?)

bubnikv, 2016-11-22:
As discussed, the models have been designed to be accepted by Simplify3D and they are as well accepted by Cura (at least the legacy one). Both S3D and Cura handle the intersecting bodies as separate entities. Slic3r tries to merge them into a single solid.

Thinking about it, maybe splitting them in Slic3r and then re-aligning them one over each other would achieve a similar effect to S3D or Cura.

vvranjek, 2016-11-24:
Hi @alexrj ,
I didn't create these files, but my brain is able to understand it :) First, this structure is designed to be printed without infill, thus the internal structure exists. I can imagine  that it is hard for a computer program to determine what is inside/outside in this design, and the program is right, there likely is none. If you ask me, the "inside" should be those thin spaces in the inner wall structures. If you have a close look, the inside structure has "double" walls, and I consider the space between them, the "inside". At least thats how I see it.

The outer one-dimensional shell is designed this way so that it is "easy" to set the outer shell width, which is very important for this print.

I know it's complex, but I am sure things can be done to make it work.

Maybe a special mode can be made, where slic3r ignores whats inside and outside and each wall is considered one-dimensional. I think it would work great for this design. Of course, infill could not exist in such environment.

bricofoy, 2016-11-27:
Hi @alexrj 

What about the file I provided ? It does print OK with cura, whereas slic3r produce incorrect infill with big holes inside.
Here is a new version of the same part, completely rebuilt from scratch using a different technique for modeling it. No luck, same result, but this time there is no manifold errors reported like with my previous one.
And again this file is printed correctly with cura.

[support7.stl.zip](https://github.com/alexrj/Slic3r/files/614783/support7.stl.zip)

bubnikv, 2016-11-28:
The model provided by @bricofoy slices fine with the Prusa3D 1.31.6 build. 

The initial report by @vvranjek relates to the airplane models, which were deliberately creates as non-manifolds and the slicing result depends on the strategy of a particular slicer handling the non-manifolds.

bricofoy, 2016-11-28:
@bubnikv , my model does not slice OK with the latest stable 1.2.9. It looks fine in the 3D preview, but if you look at the slices, there are full of holes in the infill that makes the printed part unusable. Have a look at the attached screenshot showing clearly the problem :
[image]

bubnikv, 2016-11-28:
@bricofoy Then please try with https://github.com/prusa3d/Slic3r/releases/tag/version_1.31.6

bricofoy, 2016-11-28:
@bubnikv exact same result : same empty space in the infill at the same locations : 
[image]

bubnikv, 2016-11-28:
Don't you have the "infill only where needed" feature enabled?

bricofoy, 2016-11-28:
Good catch ! That solved the problem. Sorry for the noise.
What is the purpose of this feature ? because it seems to produce incorrect results, as the part produced this way will not print because there is too much empty space inside and the upper face is not  supported

bubnikv, 2016-11-28:
http://manual.slic3r.org/expert-mode/infill-optimization