slic3r/Slic3r #3524: Feature request: Allow infill layers to be different thickness (multiples) to perimeter layers

mrjohnc, 2016-10-04:
An option to print infill layers that are multiples of the height of the perimeter layers. This would allow the exterior of the model to have the same detail but would both make the printed part stronger by having thicker layers and reduce the time needed to print the infill significantly.

Example: 

- Perimeter (the detail you see on the outside of the model) is printed at 0.1mm layer height
- Infill is printed at 0.3mm layer height and is printed ever 3rd perimeter layer, this would reduce the time needed to print infill by 66%

There would need to be some allowances for shapes that would make this process difficult e.g objects that have a very strong curve or objects with horizontal sections that were 1 layer thick.

xoan, 2016-10-04:
http://manual.slic3r.org/expert-mode/infill-optimization

`Infill every n layers` (in v1.2.9 is "Combine infill every n layers"). You need to set your layer height as you want in your perimeters, and set this parameter as a multiplier for your infill height.

mrjohnc, 2016-10-04:
Thanks @xoan this seems to be the first half of my feature request, the second half would be that the infill layers were different heights so there were no gaps between infill layers.

Perhaps I am not being clear, here is a more full example:

Perimeters set to 0.1mm layer height
Infill set to 0.3mm layer height: 

This would be the sequence of events for a set of layers in a print (the same would be true for the bottom layers as well I think but it would be solid rather than infill).

1. Printer creates perimeters with a layer height of 0.1mm
2. Printer moves up 0.1mm 
3. Printer creates the next layer of perimeters at 0.1 mm layer height
4. Printer moves up 0.1mm 
5. Printer creates the next layer of perimeters at 0.1 mm layer height
6. Printer prints infill at 0.3mm layer height so the infill bonds with the previous layer of infill
7. Printer moves up 0.1mm layer height
8. Repeat from step 1

This will mean that you will have avoided 2/3rds of the time needed to produce the infill

kefir-, 2016-10-04:
There are at least theoretical arguments to be made in favour of not making it a multiplier, precisely to increase strength. The idea is that you get fewer horizontal layers through the entire part, so cracking the part between layers should be more difficult.

xoan, 2016-10-04:
Actually, this is the behaviour of the combine infill feature.

Try it.

kefir-, 2016-10-04:
By the way, your detailed description sounds like exactly what the mentioned feature does. Try it out and inspect the resulting gcode layer by layer, I think you'll be satisfied with the result.

mrjohnc, 2016-10-04:
Hi both

Thanks very much, I don't understand which option you are referring to, there is no feature called 'combine infill' and the 'Infill every n layers' option makes it clear it misses out layers so there are gaps between them. 

Thanks again

kefir-, 2016-10-04:
Did you try it and inspect the gcode? How does it make it clear that there are gaps between the layers?

xoan, 2016-10-04:
"Infill every n layers - Will produce sparse vertical infill by skipping a set number of layers. This can be used to speed up print times where the missing infill is acceptable."

"sparse vertical" make it clear that there are no gaps between infill layers.

Please, try it before comment again :)

lordofhyphens, 2016-10-04:
#1783
https://github.com/lordofhyphens/slic3r/tree/infill-combine

mrjohnc, 2016-10-04:
Thanks very much, yes this is what I meant :) I was confused by the description in the instructions 'missing infill is acceptable' which I think is wrong because it does not miss any infill at all. Also sparse vertical is technical language so I didn't understand what it means.

Thanks again

closed 2016-10-04 by mrjohnc

lordofhyphens, 2016-10-04:
@mrjohnc Just want to make sure that your question/issue is actually addressed: the combined infill feature, as intended, is supposed to address this (thicker layers in Z for stronger parts while maintaining thinner outer perimeters). There was a bug where it hadn't. I believe it should be fixed now.

mrjohnc, 2016-10-04:
@lordofhyphens I think so but I'm probably not a good person to ask to make sure, I'm not very experienced in looking G-Code. I think the description should be rewritten in the manual, its not clear and kind of misleading, should I add that as a feature request?

lordofhyphens, 2016-10-04:
Yes you may. What I was asking about was that, experimentally  (aka
printing) it does what you want. You should be able to see the effect in
the preview as well.

mrjohnc, 2016-10-05:
@lordofhyphens something like this?

"Infill every n layers - Vertical infill will be created every n layers that is 'n * layer height' high. This function can be used to speed up print times where thicker height infill layers are acceptable. For best results keep infill layer height shorter than the nozzle diameter."

Can confirm that print infill every 2 layers works very well, however it made Slic3r crash the first few times I tried to do it with no top or bottom solid layers, it crashed as soon as I clicked 0 bottom solid layers. I can't get it to reproduce the error now.

Patola, 2016-10-05:
Hi, is that line from your diff right? It seems odd to me. What's the reason to modulus 1 to the result of the division, since it always return zero? I've made a bunch of tests with different values on this line and it always zeroes out $infill_combined_flow.

+        if ($infill_combined_flow > $flow->nozzle_diameter) {
 +          # can't put out layers bigger than the nozzle width, go as big as possible
>>>>> +          $infill_combined_flow = (($flow->nozzle_diameter / $layerm->layer->object->config->layer_height) % 1) * $layerm->layer->object->config->layer_height
          }