slic3r/Slic3r #3526: Don't start bridge from the middle of the gap

Yenya, 2016-10-05:
### Version
slic3r-1.2.9-8.fc24.x86_64

### Operating system type + version
Fedora 24

### Behavior
* When priting a bridge (e.g. "bridge torture test" object from thingiverse) with "Seam position" set to "Random", slic3r often starts printing the bridge from the middle of the gap which is to be bridged. The first trace is therefore not attached to anything, and real print starts from the next trace.

Expected results: even with "Seam position = Random", the layer should start (randomly) from a point that is not part of a bridge.

[bridge.zip](https://github.com/alexrj/Slic3r/files/511049/bridge.zip)

DrLex0, 2016-11-27:
Also just noticed this on the bridge test [model](http://www.thingiverse.com/thing:12925), here's what it looks like in gcode viewer:
[image]

The result:
[image]

Still present in the prusa3d 1.31.5 build as well.

bubnikv, 2017-02-02:
I have fixed that in the Prusa3D fork. The fix avoids placement of the seam on a bridging perimeter, even in the random seam mode. This means it will avoid bridging overhangs as well. I think this is a reasonable implementation, because a seam on an overhang perimeter is bad as well.

alexrj, 2017-03-01:
Fixed in master as well, thank you.

closed 2017-03-01 by alexrj