slic3r/Slic3r #3528: Compilation of Slic3r with Visual Studio 2013 - macro errors

negative99, 2016-10-06:
### Version
Slic3r from GitHub (1.29, 3700950).

### Operating system type + version
Windows 8.1 x64

### Behavior
I'm trying to compile Slic3r with Visual Studio 2013 following instructions given by LordOfHyphens (https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#building-with-visual-studio-2013). I've run into some problem with boost (somehow it was invisible for compiler), nevertheless I've managed to overcome them by modifying Build.PL file in xs directory. 
I'm still unable to compile slic3r. Somehow, compiling perlglue.cpp gives strange errors:

Generating script 'src\perlglue.ccs'
cl -nologo -c @"src\perlglue.ccs" -Fo"src\perlglue.obj" "src\perlglue.cpp"
perlglue.cpp
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258)
: warning C4003: not enough actual parameters for macro 'PerlSock_bind'
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258)
: error C2988: unrecognizable template declaration/definition
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258)
: error C2059: syntax error : '*'
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258)
: error C2059: syntax error : '('
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258)
: error C2091: function returns function
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258)
: error C4430: missing type specifier - int assumed. Note: C++ does not support
default-int
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258)
: error C2935: 'std::_Bind<false,void,_Fun,_Types...>' : template-class-id redef
ined as a global function
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258)
: error C2059: syntax error : ','
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1267)
: error C2065: '_Rx' : undeclared identifier
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1267)
: error C2059: syntax error : 'const'
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1267)
: error C2065: '_Types' : undeclared identifier
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1267)
: error C3861: '_Rx': identifier not found
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1267)
: error C2923: 'std::_Bind' : '_Rx' is not a valid template type argument for pa
rameter '_Ret'

(...)

Those errors are repeated few times. Have you come across such problem?

best regards
Maciek

bubnikv, 2016-10-06:
You may try to compile the Prusa3D fork
https://github.com/prusa3d/Slic3r
and then to compare the differences. Maybe there are some details I forgot to document. I know perl does some crazy stuff, it redefines basic stdlib / stdio functions with macros, and these macros have to be undefined selectively.

The prusa3d fork top level is currently not stable though, but some new stuff is coming. For example, am working on supports to make them usable, with the Simplify3D as a benchmark.

bubnikv, 2016-10-06:
Yeah, I remember, it is likely the "bind" macro defined by Perl, which replaces the "bind" word in the boost libraries all over the place. Just look for #undef bind in the prusa3d fork.

negative99, 2016-10-06:
Wow, that was fast. #undefing bind worked, thank you so much for advice!

However the compilation fails with such messages:

c:\\wperl64d\\bin\\perl.exe -MExtUtils::XSpp::Cmd -e xspp -- -t "C:/slic3r/Slic3r/xs/xsp/typemap.xspt"  "C:/slic3r/Slic3r/xs/xsp/Model.xsp"(118) : error C2146: syntax error : missing ')' before identifier 'C'

Line 118 is: Clone<TriangleMesh> raw_mesh();

what is more curious, there is another error, similar:
c:\\wperl64d\\bin\\perl.exe -MExtUtils::XSpp::Cmd -e xspp -- -t "C:/slic3r/Slic3r/xs/xsp/typemap.xspt"  "C:/slic3r/Slic3r/xs/xsp/Model.xsp"(521) : error C2146: syntax error : missing ')' before identifier 'C'

But my Model.xsp file does not have line 521, it merely has 252 lines.

Do you have any idea?

negative99, 2016-10-06:
That's interesting. The error disappeared when I commented out XSRETURN_UNDEF; in Model.xsp file. 
I almost got it working, it just fails to link boost libraries but at least it compiled:).

Thank you for your help.

I haven't mentioned it before, but I had to comment out all references to CONFESS macro in cpp files. Otherwise it would not compile.

bubnikv, 2016-10-07:
The "CONFESS" problem is caused by the XS processor, whcich generates #line
pragmas, which contain something the Visual Studio cannot digest.
This is the report I gave to the XS generator maintainer:

-----------------------

Hello.

I am trying to set up a development perl environment compiled from scratch
with Visual Studio 2013 free C/C++ compiler. I stumbled over a following
problem:

If the XS file contains an INCLUDE_COMMAND call, this line is later stored
as a comment into the generated XS.c file in a form of #line NUMBER
"filepath". Here the "filepath" will be the complete parameter of
INCLUDE_COMMAND. If the parameter contains escaped quotes, this breaks the
__FILE__ substitution in the visual studio C++ compiler.

Here is an example of a #line inserted into XS.c, which the Visual Studio
C++ has a problem with:

#line 112 "c:\\\\wperl64d\\\\bin\\\\perl.exe -MExtUtils::XSpp::Cmd -e xspp
-- -t \"D:/src-perl/Slic3r/xs/xsp/typemap.xspt\"  \"D:/src-perl
/Slic3r/xs/xsp/Print.xsp\""

If I remove the quotes from the INCLUDE_COMMAND call, the resulting form is
swallowed by the Visual Studio compiler without any problem.

#line 112 "c:\\\\wperl64d\\\\bin\\\\perl.exe -MExtUtils::XSpp::Cmd -e xspp
-- -t D:/src-perl/Slic3r/xs/xsp/typemap.xspt  D:/src-perl/Slic3r/xs/xsp/
Print.xsp"

I see the following ways to fix this problem:

1) The XS generator would remove the inner quotes from the #line comments
when targeting the Visual Studio compiler

2) The XS generator would save the intermediate file generated by the
script in  INCLUDE_COMMAND, and reference it in XS.c

Thanks for consideration,
Vojtech

-----------------------------

and a follow-up:

> How exactly does it fail?  What error message does it produce?

What fails is the substitution of the __FUNCTION__ macro, which is
 Microsoft's non-standard variant of the __func__ macro defined by the C99
standard I believe. Visual Studio will fail to compile various asserts,
which unfold the __FUNCTION__ macro. I think what happens is the Visual
Studio compiler somehow incorrectly interprets the quoting. The error
message given by the compiler is confusing at least and it was quite hard
to pinpoint it to the quotes in the #line.

negative99, 2016-10-10:
Dear Vojtech, thank you very much for your help and explanations. I've finally given up compiling whole slic3r with Visual Studio. I just couldn't overcome problem with boost libraries (they were seen by linker but still it returned unresolved external symbols in boost-system etc.). Afterwards I've found CMake file in slic3r directory which allowed me to compile libslic3r and commandline slicer.exe with Visual Studio 2013. This is good enough for me:).

Thank you once more for your help.

best regards
Maciek

bubnikv, 2016-10-10:
Maciek,

try to set the SLIC3R_STATIC environment variable before running perl Build.PL
This will instruct Slic3r to be built with a static boost library. This may help.

Also please note that the cmake built Slic3r is for SLA machines only, it does not generate g-codes.

Vojtech

negative99, 2016-10-11:
Dear Vojtech, I've finally managed to compile slic3r:). Setting SLIC3R_STATIC was indeed neccessary. Moreover, as I wrote before I had to modify the Build.PL script in xs directory because it couldn't find my boost libraries. To be honest I modified it by just putting their names and path directly:). But the correct libraries which should be linked are libboost- ones, not boost- (those aren't statically linked). 

Also, somehow the Math-PlanePath module didn't install automatically. 

But all in all I have VisualStudio 2013 compiled slic3r working!

thank you once more
Maciek

bubnikv, 2016-10-11:
Congratulations, Maciek. It would be great, if you could adjust the wiki and the Build.PL with the help of @lordofhyphens , so the instructions are complete and reproducible.

You went through quite an effort of getting Slic3r built with Visual Studio. You knew beforehand, that it will be a difficult process and that getting the Slic3r compiled with strawberry Perl will be much easier. Now, Maciek, I wonder, why did you go through such struggle and what are your intentions with the Slic3r. 

I wonder, whether you are working for one of the many Polish 3D printer manufacturer. If so, please remember, that the code is AGPLed and if you give the software to your customers, or if you use part of the software at a publicly used back-end, you are obliged to publish your code.

negative99, 2016-10-13:
Dear Vojtech, I'll add my comments to the document about compiling with Visual Studio, sure. The reason I used Visual Studio is that I wanted to compile version which could be debugged with the tool I'm well familiar with. I'm doing some research on multiextruder printing so I'll make some changes in Slic3r code and it would be good to be able to debug any problems.

I'm not working for any 3D printer manufacturer. I work and the Warsaw University of Technology. Anyway I'm familiar with Slic3r license and I'm prepared to publish code if I introduce any changes.

best regards
Maciek

closed 2016-10-13 by negative99

bubnikv, 2016-10-13:
Dear Maciek,

> I'll add my comments to the document about compiling with Visual Studio,
sure.

Thanks.

> The reason I used Visual Studio is that I wanted to compile version which
could be debugged with the tool I'm well familiar with.

Yes, that was my thinking as well. Also the debugger in Visual Studio is
superb in my experience, much better than any other freely available tool
(I don't have an experience with OSX XCode though).

> I'm doing some research on multiextruder printing so I'll make some
changes in Slic3r code and it would be good to be able to debug any
problems.

I am looking forward to your research results. Maybe this will be something
we at Prusa Research may cooperate on :-) As you may now, we are currently
commercializing a multi-material print.
http://prusaprinters.org/original-prusa-i3-mk2-multi-material-upgrade-release/

Vojtech