slic3r/Slic3r #3530: Empty layer generated

benv666, 2016-10-09:
### Version
1.30.0.25-prusa3d-release_candidate_1_3-win64

### Operating system type + version
Windows 8.1 64 bit

### Behavior
A big gap is somehow generated halfway the object, as can be seen in screenshots here:
http://i.imgur.com/UBrAOTk.png
http://i.imgur.com/Bn7wPPG.png

Reproduce: with attached STL file and config, problem occurs between Z=4.70 and Z=5.35 (which is currently filled with support structure, which shouldn't happen).

Expected: no gap / support fill, just my object at those layers with possibly -some- support, not only support material.

#### STL/Config (.ZIP) where problem occurs
[filament guide v1.zip](https://github.com/alexrj/Slic3r/files/518313/filament.guide.v1.zip)
[slic3r-config.zip](https://github.com/alexrj/Slic3r/files/518322/slic3r-config.zip)

lordofhyphens, 2016-10-09:
You need to confirm this is happening with Slic3r git master (not the
prusa3d fork).

benv666, 2016-10-09:
Sure, if you have a precompiled git-master win64 build for me?

lordofhyphens, 2016-10-09:
Win32: https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-master-183#files

Basically, if this can't be confirmed to occur with our git master, it should be brought up in whatever support channels prusa3d operates.

benv666, 2016-10-09:
Just tried your win32 binary, same problem:
http://i.imgur.com/UmihoEN.png
http://i.imgur.com/W4zMJEX.png

lordofhyphens, 2016-10-09:
Your model is broken.
[image]

This is the result with a non-broken model. 
[image]
[fixed-manifold.zip](https://github.com/alexrj/Slic3r/files/518356/fixed-manifold.zip)

Closing.

closed 2016-10-09 by lordofhyphens