slic3r/Slic3r #3532: Option present in command line missing in the GUI

Tinchus2009, 2016-10-09:
1.3.0 dev

In command line there is an option called overhang and it is described as : Experimental option to use bridge flow, speed and fan for overhangs
                        (default: yes)

Can you give me more details of its use (parameters, how it is suposed to work, etc) . This option it not present in the GUI.

lordofhyphens, 2016-10-09:
Never used it myself. If it is actually already present as a config option
all you would need to do is add it to the gui in Tab.pm.

Tinchus2009, 2016-10-09:
Ok. I will try that. Any reference documentation or por on how to do that. And more important: wich parameters does this option accept?
Thanks

lordofhyphens, 2016-10-09:
Nope, not a clue.

bubnikv, 2016-10-09:
This does exactly what is says: If Slic3r finds out that there is a part of
a perimeter hanging outside the object unsupported, it splits the perimeter
into supported and unsupported pieces. The supported pieces are printed as
usual, while the unsupported pieces are printed as bridges, that means with
round cross-section of the extrudate, not with a squarish cross-section.
The transition between the normal and bridging flow is abrupt, which is not
very good.

This feature is not well documented indeed and without this feature to be
enabled, bridges are not correctly printed over supports, if enclosed by an
overhang perimeter. It is either/or: Either the overhang perimeters are
printed with the bridging flow as well as the bridges, or the bridges shall
be printed with normal flow over supports. In both cases the supports shall
adapt to heed the gap between the object and the print.

Tinchus2009, 2016-10-09:
Ok. Im almost sure this behaviour was not present in 1.2.9, and this 
feature explains the bridge being printed in the borders of some 
overhangs I have been seeing and the cooler being activated with the set 
bridging %.

So far works somehow good, teh only comments I would like to say would be:

-If the feature can detect overhangs, it would be much much better if 
the cooling would be trigered in that area but in the previous perimeter 
before the bridging, beacuse the previous perimeter most of time is not 
completly laying on top the previous layer, many times only half of the 
extrusion is suported. So cooling here is needed.

- Also, the cooling needs to be corrected regarding its timing: I 
noticed that if I leave this to be handled by this option the fan is 
activated, yes, but somehow it is "late". By the time the fan 
accelerated, in shorts overhangs ( 20 mms long), the noozle has already 
covered like 5 mms at the time the cooling has reached the desired 
power. I have made a lot of test with overhangs, just playing with 
cooling. If I leave automatic, this "overhang" option triger my cooler 
at 100% (the set bridgin % in my case), but the result is not as good as 
if I manually set the fan to work at 100% all the time. By doing it 
manually I can perfetly print 10 degrees overhangs! If you could make 
this option to triger the fan earlier to compensate the delay would be 
awesome.

alexrj, 2016-11-22:
Actually this option is already in the GUI, just with another name: "Detect bridging perimeters" in the "Layers & Perimeters" page.

Maybe its naming should be made consistent.

alexrj, 2017-03-17:
This option is now aliased as `--detect-bridging-perimeters` in order to better match the GUI name.

closed 2017-03-17 by alexrj