slic3r/Slic3r #3533: empty layer added between raft and part

leopoletti, 2016-10-11:
### Version
1.2.9
### Operating system type + version
linuxmint 18 x86_64

### Behavior
* Slic3r add an empty layer between raft and part. Thus space between raft and part is layer_thickness+contact_Z_distance instead of only contact_Z_distance.
with a layer thickness of 0.3mm and a Z distance of 0.02, I want only one raft layer:
Z for first layer: 0.3
Z for second layer: 0.92 (should be 0.62)
[image]
[image]

If I set Z distance to 0, Z second layer is 0.6 as expected
[image]

#### STL/Config (.ZIP) where problem occurs
I have this with all STL

[config.ini.zip](https://github.com/alexrj/Slic3r/files/521893/config.ini.zip)

bubnikv, 2016-10-11:
>    <https://cloud.githubusercontent.com/assets/5619357/19271966/16b05ee6-8fc7-11e6-895f-c808a3a57c71.png> [image:
>    capture du 2016-10-11 15-25-40]

I am not sure how the raft in Slic3r works, but shall it not be a tiny gap
between the print and the raft to make the separation possible?

leopoletti, 2016-10-11:
Yes this is the reason of the contact_Z_Distance which by default 0.2. But Slic3r add also the layer thickness which lead to a gap of 0.5. This is to bing because your part will be 0.3mm taller

maukcc, 2016-11-03:
In 1.3.0 the actual Z distance is "nozzle size" + contact_Z_Distance. 
unless contact_Z_Distance is zero, then it is actually zero

leopoletti, 2016-11-08:
This leads to huge error when the nozzle is 0.8: you ve got 1mm offset so 1mm déviation for the corresponding z dim....

lordofhyphens, 2016-11-08:
There is a pull request out to remove the nozzle diameter from
consideration.

lordofhyphens, 2017-01-26:
Unfortunately removing the nozzle diameter from consideration breaks everyone's configurations. 

And 0 is special-cased to be 0 exactly. Thus, you have a "dead zone" of separation. The reason for this is bridging flow has no layer height and is assumed to be the full nozzle diameter.

lordofhyphens, 2017-01-26:
Any comments @alexrj ?

bubnikv, 2017-01-31:
For non-soluble support Slic3r prints the bottom of an object (bottom above the raft, overhangs, bridges) with a bridging flow. I believe there is a bug in the support / raft logic, where a nozzle diameter is considered instead of the bridging flow height for the calculation of the support / raft separation gap. Other than that, the logic is sound as long as the bridging flow was really used for printing the object bottom.

We may discuss whether the separation logic works well for support. One may likely want more boding to the raft than to the support tops, which would mean a new set of parameters to describe the raft interface. I will look into that, but it has not a priority.

A similar issue 
https://github.com/prusa3d/Slic3r/issues/102
refers the gap and asks for a negative gap, which would force the bridging threads into the support. This is something you would not generally want for supports as this would make the support removal in difficult to access places nearly impossible, but for some specific models it may be useful.

bubnikv, 2017-01-31:
As usual, for the raft to work you have to enable the bridging perimeters, otherwise the infills will be filled in with a bridging flow, but the perimeters not. Also please note that due to the bug in the gap calculation the raft will not work well if a bridging flow modifier is set far from unity.

If desperate, you may try the workaround described in prusa3d#102 , which allows you to set a negative gap.

mat813, 2019-01-15:
Ah, while fighting adherence issues, and trying rafts to "fix" the issue, I stumbled upon this.

DARKFiB3R, 2023-01-26:
Not sure if this is the correct solution but I was having this issue, and changing the Z-offset type from `From Filament` to `From Plane`, allowed me to close the gap.

Print Settings > Options for support material and raft > Z-offset > Type: From Plane

Print Settings > Raft > Advanced > Raft contact Z distance: 0.2mm (or whtever)