slic3r/Slic3r #3534: Setting "Top Solid Infill" to any any percentage value produces holes in top layer

Pointedstick, 2016-10-12:
### Version
1.2.9

### Operating system type + version
Mac OS 10.12 Sierra

### Behavior
When I set any percentage value for Extrusion Width -> Top Solid Infill, the top layer is printed with insufficient flow, resulting in holes.

* _Steps needed to reproduce the problem_
  * Find Extrusion Width -> Top Solid Infill and change the default value (0) to a percentage number (e.g. 98%)

* _Expected Results_: extrusion width and spacing on top solid layers are reduced to improve surface finish, but flow rate is adjusted to compensate

* _Actual Results_: extrusion width and spacing on top solid layers are indeed reduced, and surface finish improves on *most* of the top layer, but insufficient flow rate (or something like that) causes gaps in other parts of the top layer. I've tried this with various values ranging from 80% to 99%; the same thing happens every time.

Layer preview in Slic3r:
[image]

Finished prints:
[image]

The cube on the left has the default value for Extrusion Width -> Top Solid Infill (0); the one on the right has it set to 98%.

#### STL/Config (.ZIP) where problem occurs
https://www.dropbox.com/s/t0fqb1eg39x5cr7/beveled%20cube.stl
https://www.dropbox.com/s/ekyr4vhs02jjre7/Slic3r_config_bundle.ini

lordofhyphens, 2016-10-12:
Just an FYI: the % is relative to the layer height, not any other widths.
So values  < 100% are only for if you know what you are doing.

lordofhyphens, 2016-10-12:
Could you post a screenshot of the 3d preview?

Pointedstick, 2016-10-12:
3D preview of the object:
[image]

3D preview of top layer:
[image]

lordofhyphens, 2016-10-13:
You uploaded a config bundle instead of a plain config, but from what I can see you are trying to to print lines at widths of 0.196mm or 0.098mm, with a nozzle width of 0.4mm.

From the toolpath it's indeed giving you the trace widths you're asking for. I am not surprised that your printer  fails to produce traces less than half of the width of your nozzle.

Does it fail at 450% (with the 0.1mm layer height profile) or 225% (0.2mm layer height). If not, this is a case of your printer not being up to the task of what you asked of it.

lordofhyphens, 2016-10-13:
Please reopen if you are able to replicate the effect with the widths as mentioned above, and provide a regular exported config File->Export Config (Ctrl+E).

closed 2016-10-13 by lordofhyphens

Pointedstick, 2016-10-13:
All right, looks like I need a smaller nozzle.

lordofhyphens, 2016-10-13:
Smaller nozzle restricts your speed (higher pressures) and limits the max layer height as well, just FYI.

Pointedstick, 2016-10-13:
Yeah, I know.