slic3r/Slic3r #3536: Infill every N layers is not using the right infill pattern

SeanSolberg, 2016-10-16:
### Version
1.2.9

### Operating system type + version
Windows 10

### Behavior
I'm printing a cylinder shaped object that changes widths as it builds up.    I'd like to keep all the movements in a circular pattern as much as possible for speed, reduced vibration, etc.  Slic3r is choosing the rectilinear fill pattern for the infills for all of the "solid infill every n layers.  I have the Fill Pattern set to "concentric" and i have the top/bottom fill pattern set to concentric. 

[Slic3r_cylindar_infill_issue.zip](https://github.com/alexrj/Slic3r/files/531878/Slic3r_cylindar_infill_issue.zip)
[image]
[image]

_Is this a new feature request?_  Not sure if this is a bug or a new feature request

lordofhyphens, 2016-10-16:
That is intended behavior; "solid" infill is rectilinear.

If you want 100% concentric circles, make a modifier mesh of thin
rectangles and set the infill % for the mesh.

mrjohnc, 2016-11-04:
@lordofhyphens could this also be achieved by specifying a very high number of perimeters?

SeanSolberg, 2016-11-04:
Yes, that's what I did.  It applied it to the whole object, even where I
didn't need that.  However, it helped.

lordofhyphens, 2017-01-25:
This is now done using modifier meshes like I mentioned above. 1.3.0-dev now has modifier meshes  you can define and place in Slic3r itelf.

Closing as it's not actually a bug. You will want to set top/bottom solid layers to 1 as well when doing this.

closed 2017-01-25 by lordofhyphens