slic3r/Slic3r #3538: avoid crossing of non-printing mouvements and printed material

amiacu, 2016-10-18:
Slic3r version 1.2.9

Windows 10

Behavior

I work with a paste extruder witch imply swelling of the deposited material. When I print many (or big) pieces, the extruder head has non-printing movements that pass through printed material and damage it.

I would like to add some code like that in the gcode: 

G1 X154.140 Y124.877 E0.08073 ; infill
M204 S1500 ; adjust acceleration
G92 E0 ; reset extrusion distance
**_G1 Z1.200 ; avoid crossing_**
G1 X127.504 Y121.380 F4800.000 ; move to first infill point
**_G1 Z0.600 ; ajust height_**
M204 S500 ; adjust acceleration
G1 X121.620 Y115.496 E0.00679 F300.000 ; infill ...

so the paste extruder head doesn't touch the printed material.

Sure I can do that manualy but I wonder if there is an easier way.

If someone can help me I will be grateful.

Best regards

Amiacu

lordofhyphens, 2016-10-18:
Avoid crossing perimeters already tries to do that.

If your system is very sensitive to the print head dragging over previous
layers, then Z lift is for you.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

amiacu, 2016-10-18:
Thank you for yours answer. 

The thing is Z lift is available only if I retract the filament but doing that, I can induce bubbles in my paste (and I don't want it).

I'll see how it works.

pease

lordofhyphens, 2016-10-18:
Use a very very small retraction amount, like 0.1 or 0.001mm, and you
should be fine.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

closed 2016-11-22 by alexrj