slic3r/Slic3r #3539: Can't compile Slic3r on Win 7 32bit

pyforever, 2016-10-20:
### Version
_slic3r-mswin-x86-1-2-9a-stable_
_strawberry-perl-5.22.2.1-32bit_
_boost_1_59_0_

[image]

### Operating system type + version
_Windows 7 Ultimate SP1_

### Behavior
* _Followed all instructions in "https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows". All ok until "perl Build.PL": compilation aborts when "building xs.dll file from 'src\perlglue.cpp' at line 129 of ....Windows.pm"
* _Steps needed to reproduce the problem_
  * _Just follow instructions at "https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows" without any change_
* _Actual Results_
  * _Screenshots enclosed_

_Is this a new feature request?_ NO

I browsed the forum and tried to force manual installation, tried many times, deleted and rebuilt all from scratch but the result didn't change.
Please help, I'm going nut!
Thanks.

lordofhyphens, 2016-10-20:
... weird. Could you give it a shot from a checkout of the source code of master? 

My suspicion though is it's some kind of environment/PATH issue though.

pyforever, 2016-10-20:
I guessed it's a path issue but I've no idea of what to correct.
Exactly what do you mean by "checkout of the source code of master"?

lordofhyphens, 2016-10-20:
https://github.com/alexrj/Slic3r/archive/master.zip

pyforever, 2016-10-20:
Checked at "C:\dev\boost_1_59_0\boost\mpl\aux_\preprocessed\gcc\": indeed there's no file "PerlSock_bind.hpp"! But I've no idea about why it isn't there. Do you?

pyforever, 2016-10-20:
I forgot to tell that I downloaded a fresh new master.zip from the addres you told (https://github.com/alexrj/Slic3r/archive/master.zip) and tried a new buil but got the same result once again.

Tinchus2009, 2016-10-20:
I think I have the same issue: I was running slic3r and updateing it from time ti time useing GIT with no problems. Today I issued a git pull and then perl Buil.pl as always. This time I got this:

Generating main XS file...
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -fwrapv -fno-strict-aliasing -mms-bitfields -xc++ -Isrc/libslic3r -IC:\dev\boost_1_59_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 -I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o "src\perlglue.o" "src\perlglue.cpp"
In file included from C:\dev\boost_1_59_0/boost/mpl/bind.hpp:50:0,
                 from C:\dev\boost_1_59_0/boost/mpl/lambda.hpp:18,
                 from C:\dev\boost_1_59_0/boost/mpl/apply.hpp:25,
                 from C:\dev\boost_1_59_0/boost/iterator/iterator_facade.hpp:37,
                 from C:\dev\boost_1_59_0/boost/range/iterator_range_core.hpp:27,
                 from C:\dev\boost_1_59_0/boost/range/iterator_range.hpp:13,
                 from C:\dev\boost_1_59_0/boost/range/as_literal.hpp:22,
                 from C:\dev\boost_1_59_0/boost/algorithm/string/trim.hpp:19,
                 from C:\dev\boost_1_59_0/boost/algorithm/string.hpp:19,
                 from src/libslic3r/Config.hpp:13,
                 from src/xsinit.h:48,
                 from src\perlglue.cpp:2:
C:\dev\boost_1_59_0/boost/mpl/aux_/include_preprocessed.hpp:37:90: fatal error: boost/mpl/aux_/preprocessed/gcc/PerlSock_bind.hpp: No such file or directory
 #   include BOOST_PP_STRINGIZE(boost/mpl/aux_/preprocessed/AUX778076_PREPROCESSED_HEADER)
                                                                                          ^
compilation terminated.
error building xs.dll file from 'src\perlglue.cpp' at C:/Strawberry/perl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 129.
FAIL
! Installing ./xs failed. See C:\Users\Martin\.cpanm\work\1476981387.12604\build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

lordofhyphens, 2016-10-25:
Right. @alexrj just made commits yesterday and the perlglue reminded me of
the commit.

Master may just be broken currently -.-.

I suggest using build 183 from bintray.com if you just want to run recent
Slic3r.

pyforever, 2016-10-25:
Now I installed a 1.27 pre-compiled (win 7) and it seems it works.

lordofhyphens, 2016-10-25:
Yeah, that would probably work because it has all of the libs in one place
(independent from your system setup and paths and so forth). I'm sorry I
haven't had time to respond and help you figure this out; I've been stuck
in the lab when I wasn't sleeping.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Tinchus2009, 2016-10-26:
Sorry, what is the solution?

lordofhyphens, 2016-10-26:
There isn't one at present because I don't know how to reproduce his
problem.

I also don't know if the OP has tried one of the builds off of bintray.com

Tinchus2009, 2016-10-29:
Ok, this problem appeared in the last update of master like a week ago. 
I did a git pull, it downloaded some changes, tehn the perl Buil.PL gave 
the error, it is like there is some file missing or a reerenced path is 
wrong in some part of the code?

Here it is my log:

-DBOOST_ASIO_DISABLE_KQUEUE -DBOOST_LIBS -Isrc -Ibuildtmp -s -O2 
-I"C:\STRAWB~1\perl\lib\CORE" -I"C:\STRAWB~1\c\include" -o 
"src\perlglue.o" "src\perlglue.cpp"
In file included from C:\dev\boost_1_59_0/boost/mpl/bind.hpp:50:0,
                  from C:\dev\boost_1_59_0/boost/mpl/lambda.hpp:18,
                  from C:\dev\boost_1_59_0/boost/mpl/apply.hpp:25,
                  from 
C:\dev\boost_1_59_0/boost/iterator/iterator_facade.hpp:37,
                  from 
C:\dev\boost_1_59_0/boost/range/iterator_range_core.hpp:27,
                  from 
C:\dev\boost_1_59_0/boost/range/iterator_range.hpp:13,
                  from C:\dev\boost_1_59_0/boost/range/as_literal.hpp:22,
                  from 
C:\dev\boost_1_59_0/boost/algorithm/string/trim.hpp:19,
                  from C:\dev\boost_1_59_0/boost/algorithm/string.hpp:19,
                  from src/libslic3r/Config.hpp:13,
                  from src/xsinit.h:48,
                  from src\perlglue.cpp:2:
C:\dev\boost_1_59_0/boost/mpl/aux_/include_preprocessed.hpp:37:90: fatal 
error: boost/mpl/aux_/preprocessed/gcc/PerlSock_bind.hpp: No such file 
or directory
  #   include 
BOOST_PP_STRINGIZE(boost/mpl/aux_/preprocessed/AUX778076_PREPROCESSED_HEADER)
^
compilation terminated.
error building xs.dll file from 'src\perlglue.cpp' at 
C:/Strawberry/perl/lib/ExtUtils/CBuilder/Platform/Windows.pm line 129.
FAIL
! Installing ./xs failed. See 
C:\Users\Martin\.cpanm\work\1477756107.12964\build.log for details. 
Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

lordofhyphens, 2016-10-29:
Probably,  I noticed it in the auto build but have not had time to go fix
it. @alexrj introduced a build issue on win32 with a recent commit.

phizz166, 2016-11-01:
I got the same issue with a new clone of the master made earlier today. Identical error log as in pyforever's first post, down to the files and line numbers.

lordofhyphens, 2016-11-01:
As I said earlier,  looks like a commit by @alexrj broke the build. Use the
prebuilt -183 from bintray.com for now until I can fix it.

phizz166, 2016-11-01:
Does -183 include [patch 3443](https://github.com/alexrj/Slic3r/pull/3443) for the broken cooling logic? The software isn't usable without that. I am currently trying to build from source because I want both the fixed thin-wall logic from [2960](https://github.com/alexrj/Slic3r/issues/2960#issuecomment-198766557), which isn't in the release yet, and also the fixed cooling logic, which was broken sometime around April 12th.

uclaros, 2016-11-01:
The problem seems to be with some Perl macros conflicting with Boost library, as mentioned in another issue by @bubnikv and was indeed introduced in `68acf19`.
I have [a branch on my fork](https://github.com/uclaros/Slic3r/tree/perl_macro_conflict) that fixes the build by Undef-ing some macros in xs\src\xsinit.h , but haven't issued a pull request because I don't know if the undefs break something else.

lordofhyphens, 2016-11-12:
@uclaros I have a different fix for this which is getting pulled in as soon as I verify it builds on the server and doesn't break Linux :)

@alexrj added two deserialize functions for new ConfigOption classes, which use boost::lexical_cast. However, these implementations were in the header file and so he added includes for the boost libs there. Config.hpp is included everywhere and so is apparently not safe (at least with the current build system) because of the macros that @uclaros noticed. 

My workaround pushes the two affected functions into Config.cpp. I'm not an expert in XS/Perl/Boost and this build system, so it's the best that I can do here. Suggest avoiding adding Boost includes into Config.hpp unless someone figures out this restriction

lordofhyphens, 2016-11-12:
#3572 should fix this.

lordofhyphens, 2016-11-12:
It builds on the appveyor server and on a VM I spun up to check it, so things should be fine.

alexrj, 2016-11-13:
Well, both solutions look fine.
The #undef thing was already in place for some other symbols, so I already run into this in the past.
Thank you all for the report, investigation and fix.

Tinchus2009, 2016-11-13:
Im sorry to reports this: now the compilation is working ok, under win10.

But when I try to launch slic3r, it crashes, the only information 
available is : "perl interpreter has stopped working"

lordofhyphens, 2016-11-13:
@Tinchus2009 #3540 

Something's changed with OpenGL.pm, causes segfault.

Tinchus2009, 2016-11-13:
That was exactly what I was going to comment: I saw opengl being 
configured and I dont remember that was on previous versions? Anyway, I 
noticed also this: at the beggining of compilation, I saw an error 
saying CC: command not recongniced....

unfortunatly it scrolled, and I could not get a copy.

lordofhyphens, 2016-11-13:
copy gcc.exe to cc.exe.

lordofhyphens, 2016-11-14:
OpenGL.pm has always been installed as part of the GUI stuff. It's required
for the plater to run.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

uclaros, 2016-11-14:
Well @lordofhyphens maybe the ugly `#undef`s could be removed from xsinit.h then?

lordofhyphens, 2016-11-27:
Issue strikes again, because boost/thread.hpp got used in libslic3r.hpp.

lordofhyphens, 2016-11-27:
@phizz166 by the way the cooling logic should be fixed now.

lordofhyphens, 2017-02-22:
Should be fixed now. Can also use autobuilds now too.

PR #3572 (2016-11-12): Fix/workaround for Boost issues in Config.hpp
Moved new deserialize functions (and their accompanying boost dependencies) into Config.cpp from Config.hpp.

Fixes #3539
commit b8749154 (2016-11-12): Fix/workaround for Boost issues in Config.hpp (#3572)

* Moved new deserialize functions (and their accompanying boost dependencies) into Config.cpp from Config.hpp.

Fixes #3539

* Forgot to save Config.hpp for removal of Boost
closed 2017-02-22 by lordofhyphens