slic3r/Slic3r #3540: Ubuntu 16.10 segmentation fault when plater is displayed on 1.3.0-dev

mloebl, 2016-10-21:
### Version
_1.3.0-dev_

Git describe:
_1.2.9-367-g8910b49_

### Operating system type + version
_Ubuntu 16.10 x64_

### Behavior
* Segmentation fault launching with plater 
* Running in any mode that shows the plater
  * if I start with --no-plater it starts up ok
  * I think it may be an opengl related issue 
  *  Precompiled linux binary of 1.2.9 works fine 
  *  Worked fine on 16.04 before I upgraded to 16.10 and installed latest Nvidia driver 370.28 (done at same time)
  *  no errors during build, here's gui build output: 

App::cpanminus is up to date. (1.7042)
Class::Accessor is up to date. (0.34)
Growl::GNTP is up to date. (0.21)
LWP::UserAgent is up to date. (6.15)
Net::Bonjour is up to date. (0.96)
OpenGL is up to date. (0.70)
Socket is up to date. (2.024)
Wx is up to date. (0.9928)
Wx::GLCanvas is up to date. (0.09)

* End of output from trace:

>> (eval 254)[/usr/local/lib/x86_64-linux-gnu/perl/5.22.2/OpenGL.pm:6194]:1: sub OpenGL::GL_TRIANGLES { 4 }
>> /usr/share/perl5/Class/Accessor.pm:175:         my $self = shift;
>> /usr/share/perl5/Class/Accessor.pm:177:         if(@_) {
>> /usr/share/perl5/Class/Accessor.pm:180:             return $self->get($field);
>> /usr/share/perl5/Class/Accessor.pm:158:     my $self = shift;
>> /usr/share/perl5/Class/Accessor.pm:160:     if(@_ == 1) {
>> /usr/share/perl5/Class/Accessor.pm:161:         return $self->{$_[0]};

This is where it segfaults

mloebl, 2016-10-21:
Thanks for any help!

mac-miller, 2016-10-21:
I have exact the same Problem, but with  Ubuntu 16.04 and Nvidia driver 370.28.

mloebl, 2016-10-21:
Thanks for following up; I'd go back a version of the driver, but I believe there's a bug with the NVidia drivers on 16.10 prior to this version.   It's weird how the precompiled version works.

mac-miller, 2016-10-22:
I do not know anything about OpenGL-programming, but i did a litte search on the web and found this statement: 
"For only nVidia PC drivers (not OS X), if you enable a vertex array that doesn't exist in the shader, it'll crash with a access violation when you attempt to draw on the shader. Other drivers ignore this, as it's really a no-op."

So i enabled lines 762 and 774 and disabled lines 758 and 772 (glEnableClientState(GL_VERTEX_ARRAY) )  and slic3r starts with plater without any problem.

That's all i can do. Perhaps someone with opengl-programming-skills can have a look at it.

bubnikv, 2016-10-25:
It is difficult to give you any advice. If you could get the stack trace of
the segfault, we would know more.

You can also try the pre-built Prusa3D release, but with Linux there may be
some missing shared library dependencies.

https://github.com/prusa3d/Slic3r/releases/tag/version_1.30.0

Chaosbit, 2016-10-25:
Having a similar issue,
running Mint 17.3 with intel onboad card.

stack trace looks like this:
`#0  __memcpy_sse2_unaligned ()
    at ../sysdeps/x86_64/multiarch/memcpy-sse2-unaligned.S:33
#1  0x00007fffe5ede1ad in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#2  0x00007fffe5f022ce in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#3  0x00007fffe5f02bc4 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#4  0x00007fffe5f02d66 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#5  0x00007fffe5f5167a in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#6  0x00007fffe5f01597 in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#7  0x00007fffe5d5e79d in ?? () from /usr/lib/x86_64-linux-gnu/dri/i965_dri.so
#8  0x00007fffe9ea589c in XS_OpenGL_glDrawArrays (my_perl=<optimized out>, 
    cv=<optimized out>) at pogl_gl_Accu_GetM.c:1176
#9  0x00007ffff7b0ac46 in Perl_pp_entersub () from /usr/lib/libperl.so.5.18
#10 0x00007ffff7b03266 in Perl_runops_standard () from /usr/lib/libperl.so.5.18
#11 0x00007ffff7a944d0 in Perl_call_sv () from /usr/lib/libperl.so.5.18
#12 0x00007ffff33e72ad in wxPliEventCallback::Handler (this=<optimized out>, 
    event=...) at cpp/e_cback.cpp:93
#13 0x00007ffff278e966 in wxEvtHandler::ProcessEventIfMatches(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-2.8.so.0
#14 0x00007ffff278ec8f in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-2.8.so.0
#15 0x00007ffff278ed42 in wxEvtHandler::ProcessEvent(wxEvent&) ()
   from /usr/lib/x86_64-linux-gnu/libwx_baseu-2.8.so.0`

tareko, 2016-10-27:
I am having the same issue with an intel card. I was on 16.04, and thought perhaps that was responsible, so I moved to 16.10 - same result.
I'm not sure how to get a stack trace, but when I run with perl -d, I get:

Slic3r-git$ perl -d ./slic3r.pl

```
Loading DB routines from perl5db.pl version 1.49
Editor support available.

Enter h or 'h h' for help, or 'man perldebug' for more help.

main::(./slic3r.pl:17):	$|++;
  DB<1> c
07:36:52 AM: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1010,wx containers,compatible with 2.8).
Signal SEGV at /Slic3r-git/lib/Slic3r/GUI/3DScene.pm line 97.
	Slic3r::GUI::3DScene::Base::__ANON__[/Slic3r-git/lib/Slic3r/GUI/3DScene.pm:98](Slic3r::GUI::Plater::3D=HASH(0x557d6e4f5148), Wx::PaintEvent=SCALAR(0x557d6be70760)) called at ./slic3r.pl line 117
	eval {...} called at ./slic3r.pl line 117
Aborted (core dumped)
```

flannelhead, 2016-11-01:
Having built Slic3r recently, I also faced this problem on Arch Linux.

I believe this is related to the recent update of the Perl OpenGL module and is specific to the i915 Intel driver. A temporary solution would be to downgrade the module. I'm soon going to open a pull request to fix this.

uclaros, 2016-11-01:
I had the same issue on windows trying to build with strawberry perl's mingw 64 bit.
Removing OpenGL 0.70 and manually installing OpenGL 0.6703 fixed it for me.
OpenGL 0.6703 is removed from http://www.cpan.org/authors/id/C/CH/CHM/ so I installed it from some mirror I found: ftp://ftp.uni-siegen.de/pub/CPAN/modules/by-module/OpenGL/OpenGL-0.6703.tar.gz
```
cpanm -U Wx::GLCanvas
cpanm -U OpenGL
cpanm ftp://ftp.uni-siegen.de/pub/CPAN/modules/by-module/OpenGL/OpenGL-0.6703.tar.gz
cpanm Wx::GLCanvas
```

If you try to `perl Build.PL --gui` after that, OpenGL 0.70 is downloaded again, so don't.

tareko, 2016-11-05:
@flannelhead's solution worked for me as well. Thank you!

lordofhyphens, 2016-11-12:
@flannelhead it's not specific to the intel driver, I can reproduce it on Debian with the nvidia proprietary driver.

The first call that causes a crash is one to glDrawArrays around line 760 in 3DScene.pm. This is just the symptom though; I'm researching OpenGL.pm-0.70 to figure out what might be going on; it's been a long time since I've actively worked with OpenGL.

Odds are that the array that should be referenced by the call to glVertexPointer_p (which looks like a wrapper function to glVertexPointer, used to provide the correct typing) isn't giving us the a real array to copy.

I grabbed the source for OpenGL.pm-0.70 and fwiw its test suite does run, so there's some thing usage-related that has changed (I suspect).

lordofhyphens, 2016-11-13:
Currently running a git bisect with OpenGL.pm to see what changed to break things.

lordofhyphens, 2016-11-13:
Looks like commit e7bf734b1295e6196675c40374f963df6b6f9513 in OpenGL.pm is the first one that breaks things, which changed several bind calls.

I am unsure at present what (if anything) Slic3r is doing wrong. 

Here's the diff: 
````
Author: Bob Free <bfree@graphcomp.com>  2015-06-10 01:26:05
Committer: Christian Walde <walde.christian@googlemail.com>  2016-09-27 12:43:05
Parent: 0596a1796d3115070b90cef6de10ed79648174cf (improve the safety of some pointer handling)
Branches: master, remotes/origin/bleeding, remotes/origin/master, remotes/origin/questionable_glutclose_fix, remotes/origin/rel
Follows: OpenGL-0.6704
Precedes: OpenGL-0.6704_91, bleeding-0.67048

    allow all glBindBuffer uses by GL2.0 instead of ARB_vertex_buffer_object

----------------------------- pogl_gl_Accu_GetM.xs -----------------------------
index ac3ba1a..fe622e0 100644
@@ -393,15 +393,18 @@ glColorPointer_p(size, oga)
 	OpenGL::Array oga
 	CODE:
 	{
-#ifdef GL_ARB_vertex_buffer_object
+		GLvoid * data = oga->data;
+#ifdef GL_VERSION_2_0
+		glBindBuffer(GL_ARRAY_BUFFER, oga->bind);
+		data = NULL;
+#elif defined(GL_ARB_vertex_buffer_object)
 		if (testProc(glBindBufferARB,"glBindBufferARB"))
 		{
 			glBindBufferARB(GL_ARRAY_BUFFER_ARB, oga->bind);
+			data = NULL;
 		}
-		glColorPointer(size, oga->types[0], 0, oga->bind ? 0 : oga->data);
-#else
-		glColorPointer(size, oga->types[0], 0, oga->data);
 #endif
+		glColorPointer(size, oga->types[0], 0, data);
 	}
 
 #endif
@@ -793,15 +796,18 @@ glEdgeFlagPointer_p(oga)
 	OpenGL::Array oga
 	CODE:
 	{
-#ifdef GL_ARB_vertex_buffer_object
+		GLvoid * data = oga->data;
+#ifdef GL_VERSION_2_0
+		glBindBuffer(GL_ARRAY_BUFFER, oga->bind);
+		data = NULL;
+#elif defined(GL_ARB_vertex_buffer_object)
 		if (testProc(glBindBufferARB,"glBindBufferARB"))
 		{
 			glBindBufferARB(GL_ARRAY_BUFFER_ARB, oga->bind);
+			data = NULL;
 		}
-		glEdgeFlagPointer(0, oga->bind ? 0 : oga->data);
-#else
-		glEdgeFlagPointer(0, oga->data);
 #endif
+		glEdgeFlagPointer(0, data);
 	}
 
 #endif

----------------------------- pogl_gl_GetP_Pass.xs -----------------------------
index 73353aa..213e518 100644
@@ -751,15 +751,18 @@ glIndexPointer_p(oga)
 	OpenGL::Array oga
 	CODE:
 	{
[... 218 lines cut ...]
 #//# glEdgeFlagPointerEXT_c($stride, $count, (CPTR)pointer);
@@ -346,15 +351,18 @@ glEdgeFlagPointerEXT_p(oga)
 		loadProc(glEdgeFlagPointerEXT,"glEdgeFlagPointerEXT");
 	CODE:
 	{
-#ifdef GL_ARB_vertex_buffer_object
+		GLvoid * data = oga->data;
+#ifdef GL_VERSION_2_0
+		glBindBuffer(GL_ARRAY_BUFFER, oga->bind);
+		data = NULL;
+#elif defined(GL_ARB_vertex_buffer_object)
 		if (testProc(glBindBufferARB,"glBindBufferARB"))
 		{
 			glBindBufferARB(GL_ARRAY_BUFFER_ARB, oga->bind);
+			data = NULL;
 		}
-		glEdgeFlagPointerEXT(0, oga->item_count, oga->bind ? 0 : oga->data);
-#else
-		glEdgeFlagPointerEXT(0, oga->item_count, oga->data);
 #endif
+		glEdgeFlagPointerEXT(0, oga->item_count, data);
 	}
 
 #endif // GL_EXT_vertex_array
````

lordofhyphens, 2016-11-14:
I did some poking around in test.pl from OpenGL.pm and noticed that the classic vertex buffers is never tested if VBOs are available. If I forced detection of VBO availability to false, it also segfaults. 

Joy.

lordofhyphens, 2016-11-17:
I've reported this upstream. On our side we could probably rewrite the affected code to not use glCreateVertex_p
https://sourceforge.net/p/pogl/bugs/25/

lordofhyphens, 2016-11-17:
I've narrowed down the problematic call to generating the base plate (for some reason) and grid. I've modified those to use Vertex Buffer Objects instead of the way it was done before. 

There's still some bugs in it, the cut dialog causes a segfault, probably something related to line drawing.

lordofhyphens, 2016-11-18:
Alright, referenced PR should completely resolve this now.

alexrj, 2016-11-22:
Merged, thank you everybody a lot for report, investigation and fix!  (especially thanks to @lordofhyphens)

closed 2016-11-22 by alexrj