slic3r/Slic3r #3541: Not an issue just want clarification on xy compensation. Possible feature request.

DjDemonD, 2016-10-21:
I am using 32 bit windows 1.2.9, thank you for making slic3r its excellent.

How does xy compensation work? I have been trying to figure out what exactly it does? As have a few quite experienced people on reprap forum and they can't help me.

I've tried some small cubes with positive values such as 1mm and I get smaller holes and larger external dimensions. What is this meant to achieve?

What I am after is something that can apply scaling to all parts dropped onto the virtual print bed, without having to scale each part individually. This can be saved as a setting for a given material or printer to compensate for material shrinkage or a tricky calibration problem.

lordofhyphens, 2016-10-25:
Xy compensation thickens/thins (+\-) all of the walls in the xy plane.

That's what it does, it is an advanced option to fudge around holes/fits.

bubnikv, 2016-10-25:
The xy compensation is an offset applied to each slice. Outer contours are
offsetted outside, holes are offsetted inside for positive compensation
values.

alexrj, 2016-11-22:
More information is in the manual:
http://manual.slic3r.org/troubleshooting/dimension-errors

@DjDemonD, scaling is not a proper solution to plastic shrinkage. It was discussed several times and there's general consensus on this. You can achieve an automatic scaling by putting an `M92` command in your Start G-code that changes the XYZE motor steps slightly, but I don't see what scenario are you trying to handle.

closed 2016-11-22 by alexrj