slic3r/Slic3r #3542: Feature Request: Mark areas of print bed

Quyzi, 2016-10-21:
I'm sure most of us have spots on our print bed that are damaged from assorted causes. The leveling probe came loose and the print head hit the surface and damaged the PEI in my case. I've got a pretty significant bump there now where the PEI is messed up.  Would be cool if we could mark an area in the plater so we can avoid it if possible.

lordofhyphens, 2016-10-25:
Try loading an stl as the print bed area.

bubnikv, 2016-10-25:
This could be achieved by a bitmap overlay over the bed surface. Relatively
easy to implement.

alexrj, 2016-11-22:
Yeah @Quyzi, as @lordofhyphens said you can just design your print bed as a STL file and load it. You can leave holes in the bed in the areas you don't want to use.  I do the same for the bulldog clips I have on some of my printers. It works well.

closed 2016-11-22 by alexrj