slic3r/Slic3r #3543: Issue of divided toolpath

sp3d-gs, 2016-10-23:
A really useful feature in Slic3r is merging meshes of 2 adjacent solids, as shown below with 2 adjacent cuboids.
[image]

However, when say, the infill speed of either STL is varied, their infills are separated, as shown in the following picture.
[image]

This makes the joint at which both infills meet an area of mechanical weakness. Would it be possible to print the infill with a single toolpath, occasionally changing the speed of the extruder, instead of separating it into 2 regions?

I'm using Slic3r v1.2.9 on Windows 10.

Thank you!

bubnikv, 2016-10-25:
> Would it be possible to print the infill with a single toolpath,
occasionally changing the speed of the extruder, instead of separating it
into 2 regions?

It is certainly possible, but it is an additional complexity to add to the
code and I believe there are currently more pressing issues in Slic3r.

I would rather see a general merging of infill regions with similar
properties. Currently Slic3r creates separate infill areas for top / bottom
/ bottom bridging / solid infill, even if they have very similar
parameters. If the regions are small, it makes certainly sense to merge
them, using either average or minimum settings. For example, if the
extrusion width is 0.4 for top infill and 0.45 for solid infill, it makes
often sense to extrude the whole region with 0.4.

Vojtech

sp3d-gs, 2016-10-28:
Thanks for your response bubnikv. I agree with your suggestion to merge infill regions with very similar or even identical parameters. It would make each layer more cohesive. I'm at a company that 3D prints a variety of different parts. Finding out how to merge regions with identical parameters into a single toolpath when slicing would be an important improvement, as it would make the parts mechanically stronger and increase their visual quality.

Hence we'd be willing to do this and invest time in it. How much would you estimate the time and complexity of making such a modification/change is?

alexrj, 2016-11-22:
@bubnikv, merging regions with similar properties is easy since we already merge those with identical properties and it would be a matter of comparing with thresholds instead of comparing by equality.
But if the threshold is too large, people will complain...

In any case, this would not address the request contained in this issue. @sp3d-gs, can you elaborate on the usage scenario which involves the need for different speeds within the same infill area?

lordofhyphens, 2017-01-26:
@sp3d-gs ? Anything?

bubnikv, 2017-01-27:
Re merging infill regions with similar properties:
> Hence we'd be willing to do this and invest time in it. How much would you estimate the time and complexity of making such a modification/change is?

This is difficult to estimate without a deep thought (1 or more days) to propose an efficient algorithm to do that. It is really not a problem to hack something quickly, but if it then takes days to compute over a large model, it is not very useful.

What needs to be done is to prioritize the regions (top / bottom over infill) and then merge some islands, maybe cut thin tentacles from the bigger islands first, then merge these thin tentacles with the islands of another type. Not easy to do efficiently.

sp3d-gs, 2017-02-06:
Hey guys, sorry for the late reply. The usage scenario would be very general - I thought this feature would simply speed up FDM 3d printing, as it would allow separate parts (or geometries) of 1 object to be printed with its own unique printing parameters, instead being assigned the printing settings of the slowest part/geometry. I realise it is more of a convenience than a necessity. Would you think of it as a change worth looking into?