slic3r/Slic3r #3544: Missing layer in gcode

kdupke, 2016-10-25:
[like the template]
### Version
1.2.9-367-g8910b4
github Oct 23 11:50 UTC
1.3.0-dev
### Operating system type + version
openSUSE Leap 42.1

### Behavior
Layer in between is missing. Just left out.

Slicing the attached STL one layer at 27mm is missing. Other layers are fine.

When I create the object with different parameters in openSCAD I still get a missing layer but a different one.

[image]
[image]
[image]
[image]
[uploads_da_c0_39_3e_c9_tween_example3_kd2_mesh_0.25.stl.gz](https://github.com/alexrj/Slic3r/files/550159/uploads_da_c0_39_3e_c9_tween_example3_kd2_mesh_0.25.stl.gz)

kdupke, 2016-10-25:
I just figured out the original file was optimized for 0.3mm but sliced with 0.25mm. Still wonder as the object looks fine as STL both in the openSCAD view and the slic3r 3d view.

Attached a 0.25mm optimized working version.
[hood_mesh_0.25-A.stl.gz](https://github.com/alexrj/Slic3r/files/550512/hood_mesh_0.25-A.stl.gz)

Not sure this is a real object issue or a slicr issue.

bubnikv, 2016-11-21:
Looking at the original model, the STL consists of plenty of thin slices. I don't know how you created the model with OpenSCAD, but such a model has a high risk of containing extremely sharp triangles, which are notoriously difficult to process with any computational geometry algorithm. Instead of merging thin slices, you should rather calculate hulls of simple 3D objects in OpenSCAD to achieve smooth organic like transitions. I know, it is difficult.

lordofhyphens, 2017-01-25:
This is probably not a bug in Slic3r, but the model being funky instead. 
Closing for now, has been no response from @kdupke

closed 2017-01-25 by lordofhyphens