slic3r/Slic3r #3545: Add a project file containing the platter and all referenced configuration files

bubnikv, 2016-10-26:
Slic3r currently supports exporting and loading of the platter with all objects and modifier meshes, including the object specific configuration overrides. So it seems we are close to a complete project file.

Following issues would need to be addressed though:
* Storing the Print/Filaments/Printer configurations into the AMF file (maybe store just the file names? better to store the contents of the ini files IMHO)
* Storing the object transformations (scaling, rotation), not just the position.
* Storing the layer thickness table.
* Improving the speed of loading the AMF file.
* Adding compression of the AMF file.

Did I miss something?

Maybe the 3MF format from Microsoft is a better option? Isn't the AMF format copyrighted?
Should the Slic3r platter file have a suffix different from .AMF?

Following GH issues relate to this feature request:
https://github.com/alexrj/Slic3r/issues/797
https://github.com/alexrj/Slic3r/issues/1526
https://github.com/alexrj/Slic3r/issues/2556
https://github.com/alexrj/Slic3r/issues/3167

lordofhyphens, 2016-10-26:
I would be on board with this, if only to make it easier for people to
report issues :)

Adding 3MF support was on my list of things to do after I get done with my
current non-slic3r related work hell.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

kefir-, 2016-10-26:
For the same issue reporting reason given by @lordofhyphens, I think the project file should store config for Print/Filament/Printer and not just reference by name. Of course this could lead to some confusion if you've updated the settings and then load an old project with the old settings, but there could be a warning if the settings are sostenuto (with a diff?) and perhaps a timestamp showing when each was last saved. That could help to resolve conflicting configs.

lordofhyphens, 2017-01-21:
Referencing: #2811