slic3r/Slic3r #3548: Accurate mass properties of objects

jschall, 2016-10-27:
I'm interested in getting accurate mass properties (moments of inertia) for 3D-printed objects. Traditionally, you would get mass properties of bodies from a CAD suite. However, for a 3D-printed body, the CAD model does not know the exact structure. So, I'd like to add a feature to slic3r that predicts the mass properties of the printed object.

In order to do this, I need: a list of objects in the print, with each object comprising a list of all print lines intended to be part of the final object with per-line extrusion masses, line height and line width, with all coordinates relative to the object file origin and rotated into the object file reference frame.

If someone can help point me in the right direction to generate the above data, I am happy to add this feature.

jschall, 2016-10-27:
Mass properties being:
- Mass of the object
- Location of the center of mass
- Symmetric matrix representing moment of inertia of the object about the center of mass

lordofhyphens, 2016-10-27:
You may be able to figure it out from the output gcode, if you are using volumetric E coordinates.
From there you should be able to model the internal structure and figure the mass of each layer (and it's distribution). Further calculations would take the distribution of the layers for the statistics you are interested in.

Relating the individual layers to the piece as a whole is left as an exercise to the reader.

I don't think this really has a place in Slic3r itself.

lordofhyphens, 2016-10-27:
You already know the layer height and the volume and distance of the line segment.

jschall, 2016-10-28:
The output gcode doesn't have the translations and transformations to the object file origin, nor does it differentiate different objects, support material etc (granted, that can be turned off), and slic3r obviously has all of this information internally since it is able to generate and colorize the 3d preview.

lordofhyphens, 2016-10-28:
Turn on verbose gcode. You get a lot more from it. You also have the object
as oriented

lordofhyphens, 2016-10-28:
I think most of what you otherwise want can be traced from https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/Print/Object.pm

As I said, I don't think this belongs in Slic3r and so would not look favorably at getting it into the main branch.