slic3r/Slic3r #3549: Able to recompile with dependency of glibc 2.12, instead of 2.14?

ChristopherHoffman, 2016-10-27:
I have a strange question. I am trying to run slic3r from the command-line on a shared website hosting server, which unfortunately has GNU Lib C (glibc) on version 2.12. When trying to run 'slic3r --help', I get the following error message:
```
Can't load /Slic3r/lib/vrt/2a83ce75961d6d7ff47ecfde396378ce/Util.so' for module List::Util: /lib64/libc.so.6: version `GLIBC_2.14' not found (required by /Slic3r/lib/vrt/2a83ce75961d6d7ff47ecfde396378ce/Util.so) at <content>/DynaLoader.pm line 157.
 at /Slic3r/lib/std/cavaexecscript/slic3r.pl line 13
Compilation failed in require at /Slic3r/lib/std/cavaexecscript/slic3r.pl line 13.
BEGIN failed--compilation aborted at /Slic3r/lib/std/cavaexecscript/slic3r.pl line 13.
```

Since it is a shared server, I do not have control over the versions of the libraries on the server. Are those libraries which I could try to force to require a lower version of glibc, so that I can run slic3r on that server?

lordofhyphens, 2016-10-27:
You could stick a copy of the lib in the same dir. Or compile from source.

bubnikv, 2016-10-28:
> You could stick a copy of the lib in the same dir

and set the LD_LIBRARY_PATH environment variable, if my memory serves me
well.

I would recommend to recompile as well.

Vojtech

ChristopherHoffman, 2016-10-28:
I'll give recompiling a try. Would you be able to point me in the direction of the files to change, or is it more so compiling it on a system with glibc 2.12 instead of 2.14?

lordofhyphens, 2016-10-28:
Nothing should need to be changed, just build on the system. Assuming one
of Slic3r's dependencies doesn't have a hard dependency on glibc 2.14 (this
is actually quite rare), you'll be good to go.

The reason that the precompiled Slic3r needed the newer glibc is because it
was what was present on the system when @alexrj did the build/packaging.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-10-28:
@ChristopherHoffman Going to close this, but I'll leave the conversation open so it can still be posted to. 

Build instructions are on the wiki: https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

closed 2016-10-28 by lordofhyphens