slic3r/Slic3r #3550: Support is on, then bridging is off

maukcc, 2016-10-28:
### Version
_1.3.0-dev_

### Operating system type + version
Ubuntu 16.04

### Behavior
When support is on, bridging settings will be ignored.
You will have to uncheck support, then change a bridge setting and hit enter and put the support on again. 
This is with dual extruders and extruder 2 is for support

bubnikv, 2016-10-28:
Sorry, I don't understand. Could you please be more specific?

maukcc, 2016-10-28:
when you load a "Print Settings" file with "generate support material" checked, bridging parameters like "bridges speed" and  "bridge flow ratio" are ignored.

lordofhyphens, 2016-10-28:
@maukcc Please provide an example according to the contributing guidelines (config.ini, copy of the model, picture/etc of the undesired behavior). 

We're not psychic here; anything you can do, especially with pictures, to help us understand what is going on will allow us to hopefully "fix" the problem.

maukcc, 2016-10-31:
Here you go:

[image]

[image]

and the .ini file and test stl
[bridgetest.zip](https://github.com/alexrj/Slic3r/files/561306/bridgetest.zip)

alexrj, 2016-12-17:
@maukcc, in your configuration I see:

```
support_material_contact_distance = 0
```

This means that there's no gap between object and support material, thus no bridging at all. It's intended behavior.

closed 2016-12-18 by alexrj

maukcc, 2016-12-19:
Theoretically you are right. But in practice you will need more material to be extruded and less speed while extruding, when you are printing on top of support material. 

So if we can not use the bridging parameters for this, then we will need new parameters to control printspeed and flow while printing on top of support material