slic3r/Slic3r #3551: External perimeter printing speeds

nraissov, 2016-10-31:
I recently read a post (#3543) about merging meshes/outlines and wanted to try it for myself. As described, the meshes of the 2 cuboids did merge into 1 with a common outline. However, when I varied the external perimeter speed of one of the cuboids in the settings box, and exported the new gcode to the printer, its speed did not change in practice. The extruder printed the whole outline moving at the same speed. Is this intentional? If so, why?

Varying the printing speed of a portion of the solid could really speed things up!

I am using Slic3r v. 1.2.9 on win 10. 

Thanks!

bubnikv, 2016-11-11:
I think this is a bug/feature you name it. Slic3r will try to connect perimeter of an object split by multiple modifier meshes into a single perimeter, if the perimeter settings are the same. But the "same" criteria does not take the speed into account.

alexrj, 2016-11-22:
This looks to be inconsistent behavior with infill, as #3543 shows. I see three options for handling regions with different speed:

1. merge regions and change speed while printing [ideal];
2. merge regions and use only one of the two speeds (which one? random?) [current behavior];
3. keep regions distinct [consistent with infill].

The first one requires much work, and I fail to see a proper usage scenario. @nraissov, why do you need different speeds within a single perimeter loop?

lordofhyphens, 2017-01-25:
#3519 is also probably related.