slic3r/Slic3r #3553: Segmantation fault on Debian Jessie 64b

gege2b, 2016-11-01:
### Version

Current DEV ( 8910b4970a18127199e29e71be5278b7abcbdad3 )

### Operating system type + version
_What OS are you using, and state any version #s_

Debian Jessie 8.6 x64 with some packages from jessie-backports
Perl version : v5.20.2

### Behavior
* _Describe the problem_

Segfault on GUI launch

* _Steps needed to reproduce the problem_

Just run ./slic3r.pl

Here is the output of ./slic3r.pl --debug

$ ./slic3r.pl --debug
wxWidgets version wxWidgets 3.0.2, Wx version 0.9928
Data directory: /home/gege/.Slic3r
Erreur de segmentation

Nothing else.

I can slice a STL in command-line mode, it run thru all the steps and generate a good gcode file with no error, so it seems to be GUI related, because as soon as I wish to run the GUI, it hangs.

Here is the only line appearing in syslog : 
[20985.673850] perl[8449]: segfault at 0 ip 00007fcab50a6040 sp 00007ffffdb36458 error 4 in libc-2.19.so[7fcab5014000+1a1000]

If you need more infos, please ask

Cheers
gege

lordofhyphens, 2016-11-01:
This may be your OpenGL perl version.

#3540
Version

Current DEV ( 8910b49
<https://github.com/alexrj/Slic3r/commit/8910b4970a18127199e29e71be5278b7abcbdad3>
)
Operating system type + version

*What OS are you using, and state any version #s*

Debian Jessie 8.6 x64 with some packages from jessie-backports
Perl version : v5.20.2
Behavior

   - *Describe the problem*

Segfault on GUI launch

   - *Steps needed to reproduce the problem*

Just run ./slic3r.pl

Here is the output of ./slic3r.pl --debug

$ ./slic3r.pl --debug
wxWidgets version wxWidgets 3.0.2, Wx version 0.9928
Data directory: /home/gege/.Slic3r
Erreur de segmentation

Nothing else.

I can slice a STL in command-line mode, it run thru all the steps and
generate a good gcode file with no error, so it seems to be GUI related,
because as soon as I wish to run the GUI, it hangs.

Here is the only line appearing in syslog :
[20985.673850] perl[8449]: segfault at 0 ip 00007fcab50a6040 sp
00007ffffdb36458 error 4 in libc-2.19.so[7fcab5014000+1a1000]

If you need more infos, please ask

Cheers
gege

gege2b, 2016-11-01:
just saw the PR about the 0.70 version of OpenGL

And I confirm it was the cause of the segfault
I just modified the Build.PL script as stated in the PR and it worked like a charm (using OpenGL 0.6704 instead of 0.70)

closed 2016-11-01 by gege2b