slic3r/Slic3r #3554: slic3r-dev segmentatnion fault even in console mode

wefhy, 2016-11-01:
### Version 
1.2.9-367-g8910b49

### Operating system 
4.4.0-21-generic Linux Mint 18 x86_64
i5-6200 + HD 520 (no dedicated graphics, intel only)

### Behavior
At first, after building I wasn't able to run slic3r with gui, but in console it worked fine.
I noticed there were some errors(unsupported system, lxmu and openGL) while building, I solved them by installing freeglut3-dev and libxmu-dev.
I ran Build.PL --gui again and it completed without errors, but now I get this "Segmentation fault" error when I try to run slic3r(I can't even run it without GUI, but few hours ago, before installing freeglut and libxmu it was working)
perl slic3r.pl --debug says:
wxWidgets version wxWidgets 3.0.2, Wx version 0.9928

I found similar issue from year ago:
#2535 
They were talking about libGL.so from dll folder... but I don't see it in current version. 
After searching for such files in whole disk I found some in /usr/lib/x86_64-linux-gnu, /usr/lib/x86_64-linux-gnu/mesa and blender's directory.
I also see similar there #3540 
But all the issues refer to only gui problems, for me it doesn't work even in console.

lordofhyphens, 2016-11-01:
This may be related to #3540

wefhy, 2016-11-01:
I think it's similar, but not the same. For me console one doesn't work also.
If I follow these steps in 3540 which is downgrading openGL to 0.67 and run slic3r.pl, I get:
Can't locate OpenGL.pm in @INC (you may need to install the OpenGL module) (@INC contains: ```
/home/wefhy/git/Slic3r/lib /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.22.1 /usr/local/share/perl/5.22.1 /usr/lib/x86_64-linux-gnu/perl5/5.22 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.22 /usr/share/perl/5.22 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base .) at /home/wefhy/git/Slic3r/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
BEGIN failed--compilation aborted at /home/wefhy/git/Slic3r/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 120.
Compilation failed in require at /home/wefhy/git/Slic3r/lib/Slic3r/GUI.pm line 21.
BEGIN failed--compilation aborted at /home/wefhy/git/Slic3r/lib/Slic3r/GUI.pm line 21.
Compilation failed in require at (eval 86) line 1.

```
So again doesn't work. It's the same error like before installing freeglut and libxmu.

uclaros, 2016-11-02:
Does `perl slic3r.pl --no-plater` work?

wefhy, 2016-11-02:
It does work with OpenGL 0.70, but does not with OpenGL 0.6703 (after following steps given in 3540)
It looks like when I install OpenGL 0.7, it is the same error like in 3540 issue, but another with OpenGL 0.67. Something like 0.67 not seen by slic3r for me.

uclaros, 2016-11-02:
Try removing OpenGL and Wx::GLCanvas, then manually apply the changes in PR #3552 and then run `perl Build.PL --gui` again.

wefhy, 2016-11-02:
Thanks
It leads me to conclusion that it's not really what i fought.
It looks like OpenGL 0.67 doesn't install 
That's what I get when I try to follow your steps:
```
cpanm (App::cpanminus) 1.7042 on perl 5.022001 built for x86_64-linux-gnu-thread-multi
Work directory is /home/wefhy/.cpanm/work/1478100908.2220
You have make /usr/bin/make
You have LWP 6.15
You have /bin/tar: tar (GNU tar) 1.28
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
--> Working on ftp://ftp.uni-siegen.de/pub/CPAN/modules/by-module/OpenGL/OpenGL-0.6703.tar.gz
Fetching ftp://ftp.uni-siegen.de/pub/CPAN/modules/by-module/OpenGL/OpenGL-0.6703.tar.gz
-> OK
Unpacking OpenGL-0.6703.tar.gz
Entering OpenGL-0.6703
Checking configure dependencies from META.json
Checking if you have ExtUtils::MakeMaker 6.58 ... Yes (7.24)
Configuring OpenGL-0.6703
Running Makefile.PL

In order to test your GPU's capabilities, run this make under an X11 shell

Found FreeGLUT v20801

Using interface: FREEGLUT

Checking if your kit is complete...
Looks good
Generating a Unix-style Makefile
Writing Makefile for OpenGL
Writing MYMETA.yml and MYMETA.json
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have Test::More 0 ... Yes (1.302062)
Checking if you have ExtUtils::MakeMaker 0 ... Yes (7.24)
Building and testing OpenGL-0.6703
cp Array.pod blib/lib/OpenGL/Array.pod
cp Tessellation.pod blib/lib/OpenGL/Tessellation.pod
cp OpenGL.pm blib/lib/OpenGL.pm
AutoSplitting blib/lib/OpenGL.pm (blib/lib/auto/OpenGL)
cp OpenGL.pod blib/lib/OpenGL.pod
cp Config.pm blib/lib/OpenGL/Config.pm
Running Mkbootstrap for OpenGL ()
chmod 644 "OpenGL.bs"
"/usr/bin/perl" -MExtUtils::Command::MM -e 'cp_nonempty' -- OpenGL.bs blib/arch/auto/OpenGL/OpenGL.bs 644
"/usr/bin/perl" "/usr/local/share/perl/5.22.1/ExtUtils/xsubpp" -noprototypes -typemap '/usr/share/perl/5.22/ExtUtils/typemap' -typemap '/home/wefhy/.cpanm/work/1478100908.2220/OpenGL-0.6703/typemap'  OpenGL.xs > OpenGL.xsc
mv OpenGL.xsc OpenGL.c
x86_64-linux-gnu-gcc -c  -I/usr/include -I/usr/local/include -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g   -DVERSION=\"0.6703\" -DXS_VERSION=\"0.6703\" -fPIC "-I/usr/lib/x86_64-linux-gnu/perl/5.22/CORE"  -DHAVE_VER -DHAVE_FREEGLUT -DHAVE_GL -DHAVE_GLU -DHAVE_GLUT -DHAVE_GLX -DHAVE_FREEGLUT -DHAVE_FREEGLUT_H -DGL_GLEXT_LEGACY OpenGL.c
x86_64-linux-gnu-gcc -c  -I/usr/include -I/usr/local/include -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g   -DVERSION=\"0.6703\" -DXS_VERSION=\"0.6703\" -fPIC "-I/usr/lib/x86_64-linux-gnu/perl/5.22/CORE"  -DHAVE_VER -DHAVE_FREEGLUT -DHAVE_GL -DHAVE_GLU -DHAVE_GLUT -DHAVE_GLX -DHAVE_FREEGLUT -DHAVE_FREEGLUT_H -DGL_GLEXT_LEGACY gl_util.c
"/usr/bin/perl" "/usr/local/share/perl/5.22.1/ExtUtils/xsubpp" -noprototypes -typemap '/usr/share/perl/5.22/ExtUtils/typemap' -typemap '/home/wefhy/.cpanm/work/1478100908.2220/OpenGL-0.6703/typemap'  pogl_const.xs > pogl_const.xsc
mv pogl_const.xsc pogl_const.c
x86_64-linux-gnu-gcc -c  -I/usr/include -I/usr/local/include -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g   -DVERSION=\"0.6703\" -DXS_VERSION=\"0.6703\" -fPIC "-I/usr/lib/x86_64-linux-gnu/perl/5.22/CORE"  -DHAVE_VER -DHAVE_FREEGLUT -DHAVE_GL -DHAVE_GLU -DHAVE_GLUT -DHAVE_GLX -DHAVE_FREEGLUT -DHAVE_FREEGLUT_H -DGL_GLEXT_LEGACY pogl_const.c
"/usr/bin/perl" "/usr/local/share/perl/5.22.1/ExtUtils/xsubpp" -noprototypes -typemap '/usr/share/perl/5.22/ExtUtils/typemap' -typemap '/home/wefhy/.cpanm/work/1478100908.2220/OpenGL-0.6703/typemap'  pogl_gl_top.xs > pogl_gl_top.xsc
mv pogl_gl_top.xsc pogl_gl_top.c
x86_64-linux-gnu-gcc -c  -I/usr/include -I/usr/local/include -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g   -DVERSION=\"0.6703\" -DXS_VERSION=\"0.6703\" -fPIC "-I/usr/lib/x86_64-linux-gnu/perl/5.22/CORE"  -DHAVE_VER -DHAVE_FREEGLUT -DHAVE_GL -DHAVE_GLU -DHAVE_GLUT -DHAVE_GLX -DHAVE_FREEGLUT -DHAVE_FREEGLUT_H -DGL_GLEXT_LEGACY pogl_gl_top.c
pogl_gl_top.xs: In function ‘XS_OpenGL_glpcOpenWindow’:
pogl_gl_top.xs:436:16: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 2 has type ‘Display * {aka struct _XDisplay *}’ [-Wformat=]
         printf("Display open %x\n", dpy);
                ^
pogl_gl_top.xs:456:16: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 2 has type ‘XVisualInfo * {aka struct <anonymous> *}’ [-Wformat=]
         printf("Visual open %x\n", vi);
                ^
pogl_gl_top.xs:466:16: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 2 has type ‘GLXContext {aka struct __GLXcontextRec *}’ [-Wformat=]
[... 49 lines cut ...]
mv pogl_gl_Tex2_Draw.xsc pogl_gl_Tex2_Draw.c
x86_64-linux-gnu-gcc -c  -I/usr/include -I/usr/local/include -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g   -DVERSION=\"0.6703\" -DXS_VERSION=\"0.6703\" -fPIC "-I/ [... cut ...]
Can't load 'blib/arch/auto/OpenGL/OpenGL.so' for module OpenGL: blib/arch/auto/OpenGL/OpenGL.so: undefined symbol: glWindowPos4iMESA at /usr/lib/x86_64-linux-gnu/perl/5.22/DynaLoader.pm line 187.
# at test.pl line 9.
Compilation failed in require at test.pl line 9.
BEGIN failed--compilation aborted at test.pl line 9.
Makefile:1178: recipe for target 'test_dynamic' failed
make: *** [test_dynamic] Error 2
-> FAIL Installing ftp://ftp.uni-siegen.de/pub/CPAN/modules/by-module/OpenGL/OpenGL-0.6703.tar.gz failed. See /home/wefhy/.cpanm/work/1478100908.2220/build.log for details. Retry with --force to force install it.
```
I can't track where the error comes up :/

uclaros, 2016-11-02:
Try forcing OpenGL installation with `cpanm -f` since it's a test that fails.

wefhy, 2016-11-02:
Yeah, finally it works :) 
Thanks much for your help
I'll try to remember this trick for future.
Issue solved for now.

closed 2016-11-02 by wefhy