slic3r/Slic3r #3555: Compilation issues with Slic3r 1.29.0

bwbeall, 2016-11-01:
### Version
1.29.0

### Operating system type + version
Windows 10.0.14393 Pro x64
VS 2013 Version 12.0.40629.00 Update 5
NET Framework 4.6.01586

### Behavior
Compiler errors

I've tried building version 1.29.0 using Strawberry, CitrusPerl, and VS2013 Community Edition, and have had limited success with all three.

So, I've uninstalled all of the development tools (except VS2013) and started again (with VS2013), following the instruction in Running Slic3r from git on Windows. I'll document what I've found so far, and any assistance would be greatly appreciated. First, thanks for all the effort put into this project, including the build instructions.

So, here are my build notes/questions from the VS 2013 portion.
1. How do you move the cpan config.pm from the Users folder to a global place (or more correctly, how do you let cpan know where to look for it)? I specified the other folders per your suggestions, but moving the config.pm file just had me go thru the configuration steps again.
2. When I installed cpanminus, I was informed that I could update to a newer version - any suggestions? I've found that building Slic3r is very sensitive to versions.
3. I got a bunch of errors when I try to run "perl Build.PL". There is a comment that this may be due to the boost library not being configured yet, so I'll came back to this.
4. I would suggest clarifying the instructions for IO-stringy-2.1.1 go to to the latest (earliest? - I had three for some reason) cpanm\work\XXXXXXXXXXX.XXXXX folder to run the nmake install.
5. In step 6), I think the command should be "perl Build.PL --gui"
6. Building the wxWidgets 3.0.3.beta1, I needed to make two changes: first, I added TARGET_CPU=X64; and second, I had to copy (probably could've renamed) folder vcc_x64_dll to vcc_dll.
7. I couldn't find OpenGL-0.6704 at the link provided, so I used https://github.com/zmughal/pogl/tree/OpenGL-0.6704. I'm not a big user of Cygwin, so patch is new to me (patch source_file patch_file seems to be sufficient), and I got error messages on three of the four patches I applied. I looked at the patch file (basically a diff file), and the source files, and it looked like the changes were applied properly. Also, the freeglut_static.lib should be copied to freeglut_static.lib, (not freeglut.lib). Minor nit, but there needs to be a '\' between GL and *.h on the other copy.
8. I do even less sed than Cygwin, and I couldn't get that script to work, so I did the changes manually.
9. I installed Boost 1.59.0 (based on comments in the "generic" Windows instructions), so minor tweaks were needed to the cmdline_init.bat file.
10. Back to the problems with "perl Build.PL".
11. As Vojtech noted in problem git issue #3528 (Compilation of Slic3r with Visual Studio 2013 - macro errors), __func__ needs to be replaced with __FUNCTION__ in libslic3r.h. The root problem apparently is that __func__ is no longer part of the C11 standard. I read there that the cmake build only works for SLA - I don't recall seeing any indication of this in the build instructions. I am trying to build using all three tools, so it's not a big deal, but it would be nice to know that up front (and any similar limitations on the other builds).
12. I was having issues with two erros in Config.hpp. First, some of the deserialize() methods were attempting to return the result of the operator>> (others simply returned "true"). In my code, I returned (for instance) iss.good(). The other problem was with ::max() - there is a solution provided at http://www.suodenjoki.dk/us/archive/2010/min-max.htm that seems to work (with a minor tweak):

...
#include <myinit.h>
#include "Point.hpp"

//new code from http://www.suodenjoki.dk/us/archive/2010/min-max.htm
#include <algorithm>
#ifndef max
#define max(a,b) (((a) > (b)) ? (a) : (b))
#endif
#ifndef min
#define min(a,b) (((a) < (b)) ? (a) : (b))
#endif

//#include <gdiplus.h>  // this causes all sorts of issues since this is a header, so I commented it out
#undef max
#undef min

using namespace std; // may not be needed
//end of new code

namespace Slic3r {

typedef std::string t_config_option_key;
typedef std::vector<std::string> t_config_option_keys;
...

13. BoundingBox.cpp has the same issue with ::max() that Config.hpp had, so I applied the same fix (above).
14. So, now I'm to the point where I can't seem to move forward with my VS2013 build. I'm getting a bunch of errors with the following command line:
c:\wperl64r\bin\perl.exe Build.PL --force

cl -nologo -c @"src\libslic3r\Print.ccs" -Fo"src\libslic3r\Print.obj" "src\libslic3r\Print.cpp"
Print.cpp
s:\open_source_software\slic3r\build_1_2_9\slic3r-1.2.9\xs\src\libslic3r\Point.hpp(34) : warning C4244: 'initializing' : conversion from '__int64' to 'coord_t', possible loss of data
s:\open_source_software\slic3r\build_1_2_9\slic3r-1.2.9\xs\src\libslic3r\Config.hpp(71) : warning C4244: 'return' : conversion from 'const double' to 'float', possible loss of data
s:\open_source_software\slic3r\build_1_2_9\slic3r-1.2.9\xs\src\libslic3r\Flow.hpp(35) : warning C4244: 'return' : conversion from 'double' to 'coord_t', possible loss of data
s:\open_source_software\slic3r\build_1_2_9\slic3r-1.2.9\xs\src\libslic3r\Flow.hpp(38) : warning C4244: 'return' : conversion from 'double' to 'coord_t', possible loss of data
s:\open_source_software\slic3r\build_1_2_9\slic3r-1.2.9\xs\src\libslic3r\ExPolygon.hpp(110) : warning C4267: 'return' : conversion from 'size_t' to 'unsigned int', possible loss of data
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258) : warning C4003: not enough actual parameters for macro 'PerlSock_bind'
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258) : error C2988: unrecognizable template declaration/definition
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258) : error C2059: syntax error : '*'
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258) : error C2059: syntax error : '('
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258) : error C2091: function returns function
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258) : error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\INCLUDE\functional(1258) : error C2935: 'std::_Bind<false,void,_Fun,_Types...>' : template-class-id redefined as a global function
...
and a hundred more lines like it

I can rebuild ten times in a row, and I'll get similar resuts for each build (with different, apparently random, file each time) (BridgeDetection, ExtrusionEntity, MotionPlanner, LayerRegion, PrintObject, Geometry, ClipperUtils, Polygon, ExPolygon, etc. These are all errors in the VS2012 functional file, not in the Slic3r files, so I feel that it's something not properly configured. Any suggestions?

Once I get the VS2013 build under control, I'll retry the Strawberry and CitrusPerl builds.

Thanks.

bwb

lordofhyphens, 2016-11-01:
>    archive/2010/min-max.htm that seems to work (with a minor tweak):
>    1. BoundingBox.cpp has the same issue with ::max() that Config.hpp
>    2. So, now I'm to the point where I can't seem to move forward with my

Have you read the document written by @bubnikv on the wiki?

bwbeall, 2016-11-01:
All of my comments reference "Running Slic3r from git on Windows", if that's the document you mean.

bwbeall, 2016-11-02:
Also, it was necessary to add the following to xs\src\admesh\util.c

#define _USE_MATH_DEFINES  // from https://msdn.microsoft.com/en-us/library/4hwaceh6(v=vs.110).aspx

lordofhyphens, 2016-11-08:
#3539 ?

bubnikv, 2016-11-08:
> us/library/4hwaceh6(v=vs.110).aspx

If you have the build environment set up for Visual Studio 2013, does
Slic3r Prusa Edition build?
https://github.com/prusa3d/Slic3r

lordofhyphens, 2016-11-08:
I have added the opengl fix to my fork, fyi

bwbeall, 2016-11-08:
I tried the Prusa Edition build, but it (perl Build.PL) failed due to x86 x64 conflict:
Finished generating code
Generating script 'buildtmp\XS.lds'
link @"buildtmp\XS.lds" -out:"blib\arch\auto\Slic3r\XS\XS.dll"
libboost_thread-vc140-mt-s-1_59.lib(thread.obj) : fatal error LNK1112: module machine type 'X86' conflicts with target machine type 'x64'

The build log has the following:
cpanm (App::cpanminus) 1.7042 on perl 5.024000 built for MSWin32-x64-multi-thread
Work directory is c:\wperl64r\cpanm/work/1478626279.16908
You have make "C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\BIN\amd64\nmake.exe"
You have LWP 6.15
You have C:\cygwin64\bin\tar.exe, C:\cygwin64\bin\gzip.exe and C:\cygwin64\bin\bzip2.exe
You have C:\cygwin64\bin\unzip.exe
--> Working on ./xs
Entering S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Checking if you have Module::Build 0.38 ... Yes (0.4220)
Configuring S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have Test::More 0 ... Yes (1.302062)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have Module::Build 0.3601 ... Yes (0.4220)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See C:\wperl64r\cpanm\work\1478626279.16908\build.log for details. Retry with --force to force install it.

bwbeall, 2016-11-08:
I downloaded the pre-compiled version of the Boost libraries (lib64-msvc-12.0), and it built (perl Build.PL and perl Build.PL --gui) successfully. However, when I try to run (in graphics mode - perl slic3r.pl --gui) I get the following error:

Can't locate OpenGL.pm in @INC (you may need to install the OpenGL module) (@INC contains: S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib c:/wperl64r/site/lib/MSWin32-x64-multi-thread c:/wperl64r/site/lib c:/wperl64r/lib .) at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 128.
BEGIN failed--compilation aborted at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 128.
Compilation failed in require at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI.pm line 21.
BEGIN failed--compilation aborted at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI.pm line 21.
Compilation failed in require at (eval 103) line 1.

bwbeall, 2016-11-09:
Browsing thru the OpenGL-0.7.0 folder, there is a file titled KNOWN_PROBLEMS. Bullet #2 is "No way to specify/override the compile and link options which causes problems on systems with multiple OpenGL libraries installed (e.g., 32-bit and 64-bit)." This appears to be my situation, and I'm not sure how to address it. I had both VS2013 and VS2015 installed, and have just uninstalled VS2015 (no effect). I also installed pre-compiled FreeGLUT libraries (no effect). I don't think there is an option to install only 32-bit or 64-bit tools/libraries, so, I'm not quite sure what to do next.

There are precompiled OpenGL libraries (OpenGL.ppd), and I'm trying to get those installed, but I need to add PAR-Dist-InstallPPD, which needs PAR-Dist-FromPPD, which needs Expat, which failed to install (and typically isn't needed). So I'm trying to get Expat installed, and not getting very far. I downloaded the latest source files, and build the libraries. I put those and the headers on the C: drive, and updated the Makefile to point to them. 

However, when I run nmake, I get the following (and the Expat.dll does not show up):

C:\wperl64r\.cpan\sources\authors\id\T\TO\TODDR\XML-Parser-2.44>nmake

Microsoft (R) Program Maintenance Utility Version 12.00.21005.1
Copyright (C) Microsoft Corporation.  All rights reserved.

cp Parser/Encodings/iso-8859-5.enc blib\lib\XML\Parser\Encodings\iso-8859-5.enc
cp Parser/Encodings/iso-8859-9.enc blib\lib\XML\Parser\Encodings\iso-8859-9.enc
cp Parser/Encodings/iso-8859-3.enc blib\lib\XML\Parser\Encodings\iso-8859-3.enc
cp Parser/Encodings/koi8-r.enc blib\lib\XML\Parser\Encodings\koi8-r.enc
cp Parser/Encodings/iso-8859-8.enc blib\lib\XML\Parser\Encodings\iso-8859-8.enc
cp Parser/Encodings/Japanese_Encodings.msg blib\lib\XML\Parser\Encodings\Japanese_Encodings.msg
cp Parser/Encodings/euc-kr.enc blib\lib\XML\Parser\Encodings\euc-kr.enc
cp Parser/Encodings/windows-1250.enc blib\lib\XML\Parser\Encodings\windows-1250.enc
cp Parser/Encodings/big5.enc blib\lib\XML\Parser\Encodings\big5.enc
cp Parser/Encodings/README blib\lib\XML\Parser\Encodings\README
cp Parser/Encodings/ibm866.enc blib\lib\XML\Parser\Encodings\ibm866.enc
cp Parser/Encodings/iso-8859-7.enc blib\lib\XML\Parser\Encodings\iso-8859-7.enc
cp Parser/Encodings/windows-1251.enc blib\lib\XML\Parser\Encodings\windows-1251.enc
cp Parser.pm blib\lib\XML\Parser.pm
cp Parser/Encodings/iso-8859-2.enc blib\lib\XML\Parser\Encodings\iso-8859-2.enc
cp Parser/Encodings/iso-8859-4.enc blib\lib\XML\Parser\Encodings\iso-8859-4.enc
cp Parser/Encodings/windows-1255.enc blib\lib\XML\Parser\Encodings\windows-1255.enc
cp Parser/Encodings/x-euc-jp-jisx0221.enc blib\lib\XML\Parser\Encodings\x-euc-jp-jisx0221.enc
cp Parser/Encodings/x-sjis-jisx0221.enc blib\lib\XML\Parser\Encodings\x-sjis-jisx0221.enc
cp Parser/Style/Tree.pm blib\lib\XML\Parser\Style\Tree.pm
cp Parser/Encodings/x-sjis-cp932.enc blib\lib\XML\Parser\Encodings\x-sjis-cp932.enc
cp Parser/Style/Debug.pm blib\lib\XML\Parser\Style\Debug.pm
cp Parser/Style/Objects.pm blib\lib\XML\Parser\Style\Objects.pm
cp Parser/Encodings/x-euc-jp-unicode.enc blib\lib\XML\Parser\Encodings\x-euc-jp-unicode.enc
cp Parser/Encodings/x-sjis-jdk117.enc blib\lib\XML\Parser\Encodings\x-sjis-jdk117.enc
cp Parser/Style/Stream.pm blib\lib\XML\Parser\Style\Stream.pm
cp Parser/Encodings/windows-1252.enc blib\lib\XML\Parser\Encodings\windows-1252.enc
cp Parser/LWPExternEnt.pl blib\lib\XML\Parser\LWPExternEnt.pl
cp Parser/Encodings/x-sjis-unicode.enc blib\lib\XML\Parser\Encodings\x-sjis-unicode.enc
cp Parser/Style/Subs.pm blib\lib\XML\Parser\Style\Subs.pm
        nmake -f Makefile all -nologo LIBPERL_A="libperl.lib" LINKTYPE="dynamic" OPTIMIZE="-O1 -MD -Zi -DNDEBUG -GL -fp:precise" PREFIX="c:\wperl64r\site" PASTHRU_DEFINE=" " PASTHRU_INC=" "
cp Expat.pm ..\blib\lib\XML\Parser\Expat.pm
Running Mkbootstrap for Expat ()
        "C:\wperl64r\bin\perl.exe" -MExtUtils::Command -e chmod -- 644 "Expat.bs"
        "C:\wperl64r\bin\perl.exe" -MExtUtils::Command::MM -e cp_nonempty -- Expat.bs ..\blib\arch\auto\XML\Parser\Expat\Expat.bs 644
        "C:\wperl64r\bin\perl.exe" "C:\wperl64r\site\lib\ExtUtils\xsubpp" -noprototypes -typemap C:\wperl64r\lib\ExtUtils\typemap -typemap C:\wperl64r\.cpan\sources\authors\id\T\TO\TODDR\XML-Parser-2.44\Expat\typemap  Expat.xs > Expat.xsc
        "C:\wperl64r\bin\perl.exe" -MExtUtils::Command -e mv -- Expat.xsc Expat.c
        cl -c  -I"C:\Program Files\Common Files\MSVC\Expat\include"  -nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY -O1 -MD -Zi -DNDEBUG -GL -fp:precise    -DVERSION=\"2.44\"  -DXS_VERSION=\"2.44\"  "-IC:\wperl64r\lib\CORE"   Expat.c
Expat.c
Expat.xs(275) : warning C4018: '<' : signed/unsigned mismatch
Expat.xs(341) : warning C4267: '=' : conversion from 'size_t' to 'int', possible loss of data
Expat.xs(397) : warning C4267: 'function' : conversion from 'size_t' to 'int', possible loss of data
Expat.xs(430) : warning C4244: 'function' : conversion from '__int64' to 'I32', possible loss of data
Expat.xs(439) : warning C4244: '=' : conversion from 'IV' to 'int', possible loss of data
Expat.xs(444) : warning C4244: '=' : conversion from '__int64' to 'int', possible loss of data
Expat.xs(483) : warning C4101: 'pcontext' : unreferenced local variable
[... cut ...]
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\BIN\amd64\nmake.EXE"' : return code '0x2'
Stop.

So, to summarize:
I need the OpenGL.pm, but can't install it.
There is a precompiled OpenGL.ppd, but I need a means to convert it so that I can install it.
I can't convert the PPD file because I need the expat installed, but can't compile that.

Any suggestions on:
How to get expat installed, so that I can get PAR-Dist-InstallPPD and PAR-Dist-FromPPD installed.
Or
Some other way to get OpenGL.pm installed.

lordofhyphens, 2016-11-10:
Expat is an xml library, if you aren't using AMF you can ignore it.

You can probably get it to compile. There are instructions available on the
source forge project site.

bwbeall, 2016-11-10:
In regards to your comment regarding the SourceForge Expat project, I took the latest release (2.2.0) dated 2016.06.21, which includes the instructions for building the libraries (libexpat.lib, etc.) but not for installing the XML::Parser.

To recap, my (current) critical need is OpenGL.pm

I believe that I have only two ways of getting it.
1. Using cpanm to install OpenGL.0.7.0.tar.gz
or
2. Using PAR-Dist_InstallPPD to install OpenGL.ppd

I first tried to compile OpenGL.0.7.0, but it fails, apparently because of a conflict between 32-bit source and 64-bit target. There is a note in a file (KNOWN_DEFECTS) which appears to state that there is no way to specify the correct file.

Since I reached a dead-end trying to compile OpenGL.0.7.0, I have tried to install OpenGL.ppd using PAR-Dist-InstallPPD, which depends on PAR-Dist-FromPPD, which depends on XML::Parser. So, regardless of AMF, I need XML::Parser. I followed the instructions provided by the PAR-Dist-InstallPPD installation notes to install the libexpat files (see below).

...
Expat must be installed prior to building XML::Parser and I can't find
it in the standard library directories. Install 'expat-devel' (or
'libexpat1-dev') package with your OS package manager. See 'README'.

Or you can download expat from:

http://sourceforge.net/projects/expat/

If expat is installed, but in a non-standard directory, then use the
following options to Makefile.PL:

    EXPATLIBPATH=...  To set the directory in which to find libexpat

    EXPATINCPATH=...  To set the directory in which to find expat.h

For example:

    perl Makefile.PL EXPATLIBPATH=/home/me/lib EXPATINCPATH=/home/me/include

Note that if you build against a shareable library in a non-standard location
you may (on some platforms) also have to set your LD_LIBRARY_PATH environment
variable at run time for perl to find the library.
...

I have successfully built the libraries using the instructions provided for VS2013. A few minor modifications were needed to build for a 64-bit Release target (rather than 32-bit Debug), which I believe I did correctly. I also had some issues with the EXPATLIBPATH and INCPATH in the Makefile, but believe I addressed them (missing quotes around paths with spaces), and I added the path for libexpat.dll to %PATH% (which should be equivalent to the LD_LIBRARY_PATH for Windows).

In my previous comment, I pasted the result of the link command (which should have generated the Expat.dll file, but did not), and the subsequent lib command (which should have generated the Expat.lib file, but did not) with 39 unresolved externals. It seems obvious that the first problem caused the second problem, but I don't know why the Expat.dll was not created, nor why there isn't any indication of an error.

lordofhyphens, 2016-11-10:
Welcome to CPAN by the way :(

Expat is a c library used by XML::Parser (which is a perl package). The
internals are akin to black magic to me at present.

However, that library has nothing to do with OpenGL; it is used for AMF
support.

I need to do further research, but this isn't a problem I have come across
myself and I have no idea how to reproduce it right now.

I do know that our win32 build server still works (I do need to figure out
exactly what in the last commit conflicts with boost on Windows), so this
fresh hell you are dealing with seems limited to your own system.

For VS2013, @bubnikv is a better resource;  I haven't touched it at all
since probably my undergraduate days.

bwbeall, 2016-11-10:
Thanks for the welcome - I really appreciate your help on this. I suspect this fresh hell is related to any 64-bit system (rather than just mine); as I'm searching for answers, I find a lot of this 32/64-bit nonsense.

bwbeall, 2016-11-10:
To try to figure this out, I copied the command line that failed in the NMAKE build, and pasted it to my command line. 

link -out:..\blib\arch\auto\XML\Parser\Expat\Expat.dll -dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64r\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02" Expat.obj   "C:\wperl64r\lib\CORE\perl524.lib" "A\Bunch\Of\System.Libs" -def:Expat.def -verbose

Note this command expects you to be in the Expat folder.

My experience with these tools is very limited, but here's what I think the command line should be doing.
-out: should generate a file "..\blib\arch\auto\XML\Parser\Expat\Expat.dll"
-dll should generate a DLL
-nologo means no logo
-nodefaultlib means to skip searching all default libraries for unresolved externals
-debug means generate debug info
-opt:ref means optimize unreferenced functions and data out
-opt:icf means optimize identical COMDAT folding (???)
-ltcg means the linker is to call the compiler to perform whole program optimization
-libpath: specifies the linker should search "c:\wperl64r\lib\CORE" before the environment lib path
-machine:AMD64 specifies the 64-bit architecture
-subsystem:console "5.02" means this is Win32 character mode application, with a minimum version of 5.02
Expat.obj is the name of the (only) input file to this library
XXXX.lib are the input libraries to be searched for unresolved externals - only one is not a default library
-def:Expat.def means that the linker is to use Expat.def as the Module Defintion File for this link:
LIBRARY "Expat"
EXPORTS
  boot_XML__Parser__Expat
-verbose means display verbose messaging (I added this).

Now, by deleting the nologo, debug, nodefaultlib and all of the explicit default libs, this shortens down to:

link -out:..\blib\arch\auto\XML\Parser\Expat\Expat.dll -dll -opt:ref,icf -ltcg  -libpath:"c:\wperl64r\lib\CORE" -machine:AMD64 -subsystem:console,"5.02" Expat.obj   "C:\wperl64r\lib\CORE\perl524.lib" -def:Expat.def -verbose

From all of this, I believe that what should happen is the linker should create a DLL from the Expat.obj file, resolving externals using the default and specific libs.

With this command line, I get the same 39 unresolved externals, all in the library created in the previous step (Expat.lib). So I added the Expat.lib to the list:

link -out:..\blib\arch\auto\XML\Parser\Expat\Expat.dll -dll -opt:ref,icf -ltcg  -libpath:"c:\wperl64r\lib\CORE" -machine:AMD64 -subsystem:console,"5.02" Expat.obj "C:\wperl64r\lib\CORE\perl524.lib" "..\blib\arch\auto\XML\Parser\Expat\Expat.lib" -def:Expat.def -verbose

However, when I paste this to the command line, I get the following error:

...
Finished searching libraries

Generating .exp file
LINK : fatal error LNK1149: output filename matches input filename 'C:\wperl64r\.cpan\sources\authors\id\T\TO\TODDR\XML-Parser-2.44\blib\arch\auto\XML\Parser\Expat\Expat.lib'
c:>

This appears to be generating EXP/LIB files, rather than a DLL file (although an empty DLL does appear in the same folder as the existing EXP/LIB files).

So, if I don't include the Expat.lib file in the DLL link command, there are unresolved externals. If I do include the Expat.lib file, it complains that it alredy exists.

bwbeall, 2016-11-13:
Well, I've made some progress. I was able to install OpenGL-0.70 (I was trying to install 0.6704 last week, but somehow got 0.70 installed this morning. Is that an issue?

In any case, I was able to successfully build the Prusa3D version of the code. All tests were successful:
...
All tests successful.
Files=32, Tests=457, 96 wallclock secs ( 0.16 usr +  0.06 sys =  0.22 CPU)
Result: PASS

There were three sets of tests that were silpped (dynamic.t, loops.t, and slice.t)

I built the --gui, with the following status:
App::cpanminus is up to date. (1.7042)
Class::Accessor is up to date. (0.34)
Growl::GNTP is up to date. (0.21)
LWP::UserAgent is up to date. (6.15)
Net::Bonjour is up to date. (0.96)
OpenGL is up to date. (0.70)
Socket is up to date. (2.024)
Win32::API is up to date. (0.84)
Win32::TieRegistry is up to date. (0.30)
Wx is up to date. (0.9928)
Wx::GLCanvas is up to date. (0.09)

But, when I run perl slic3r.pl, Perl crashes. When I run the debugger, I get the following output and stack info:

'perl.exe' (Win32): Loaded 'C:\wperl64r\bin\perl.exe'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\ntdll.dll'. Cannot find or open the PDB file.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\kernel32.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\KernelBase.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\bin\perl524.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\msvcr120.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\user32.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\win32u.dll'. Cannot find or open the PDB file.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\gdi32.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\gdi32full.dll'. Cannot find or open the PDB file.
'perl.exe' (Win32): Loaded 'C:\Windows\WinSxS\amd64_microsoft.windows.common-controls_6595b64144ccf1df_6.0.14393.447_none_42191651c6827bb3\comctl32.dll'. Cannot find or open the PDB file.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\msvcrt.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\combase.dll'. Cannot find or open the PDB file.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\ucrtbase.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\rpcrt4.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\bcryptprimitives.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\imm32.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\lib\auto\Cwd\Cwd.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\lib\auto\Win32\Win32.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\advapi32.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\sechost.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\ole32.dll'. Cannot find or open the PDB file.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\version.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\site\lib\auto\List\Util\Util.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\lib\auto\Fcntl\Fcntl.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\lib\auto\POSIX\POSIX.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\lib\auto\mro\mro.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\site\lib\auto\threads\threads.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\lib\auto\threads\shared\shared.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\lib\auto\attributes\attributes.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\site\lib\auto\Wx\Wx.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\msvcp120.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib\wxbase30u_vc_custom.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib\wxmsw30u_core_vc_custom.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\wperl64r\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib\wxmsw30u_adv_vc_custom.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\shell32.dll'. Cannot find or open the PDB file.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\winmm.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\cfgmgr32.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\winmmbase.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\windows.storage.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\comdlg32.dll'. Cannot find or open the PDB file.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\powrprof.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\SHCore.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\shlwapi.dll'. Symbols loaded.
'perl.exe' (Win32): Loaded 'C:\Windows\System32\kernel.appcore.dll'. Symbols loaded.
[... 94 lines cut ...]
 	wxmsw30u_core_vc_custom.dll!wxWindow::HandleSize(int __formal, int __formal, unsigned int wParam) Line 5300	C++
 	wxmsw30u_core_vc_custom.dll!wxWindow::MSWHandleMessage(__int64 * result, unsigned int message, unsigned __int64 wParam, __int64 lParam) Line 2797	C++
 	wxmsw30u_core_vc_custom.dll!wxWindow::MSWWindowProc(unsigned int message, unsigned __int64 wParam, __int64 lParam) Line 3681	C++
[... cut ...]
 	perl524.dll!Perl_pp_entersub(interpreter * my_perl) Line 3987	C
 	perl524.dll!Perl_runops_standard(interpreter * my_perl) Line 41	C
 	perl524.dll!S_run_body(interpreter * my_perl, long oldscope) Line 2485	C
 	perl524.dll!perl_run(interpreter * my_perl) Line 2406	C
 	perl524.dll!RunPerl(int argc, char * * argv, char * * env) Line 252	C++
 	[External Code]	

I've also attached a screen dump of the debugger. 
[image]

Any thoughts? Please let me know if you have any questions or need additional info.

lordofhyphens, 2016-11-13:
The segfault is #3540. I suspect that the pointer being used with glDrawArrays is no longer compatible for some reason. Please discuss segfault there.

For your current purposes, suggest using OpenGL.pm-0.58 instead

bwbeall, 2016-11-14:
FWIW, I tried OpenGL.pm 0.58, 0.6703, 0.6704, and 0.70. OpenGL 0.58 doesn't build properly:
S:\Open_Source_Software\Slic3r\Slic3r-Prusa3D_1.31.2>perl slic3r.pl --gui
Bareword "GLU_TESS_BEGIN" not allowed while "strict subs" in use at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 395.
Bareword "GLU_TESS_END" not allowed while "strict subs" in use at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 396.
Bareword "GLU_TESS_VERTEX" not allowed while "strict subs" in use at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 397.
Bareword "GLU_TESS_COMBINE" not allowed while "strict subs" in use at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 398.
Bareword "GLU_TESS_ERROR" not allowed while "strict subs" in use at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 399.
Bareword "GLU_TESS_EDGE_FLAG" not allowed while "strict subs" in use at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 400.
BEGIN not safe after errors--compilation aborted at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI/Plater/2DToolpaths.pm line 881.
Compilation failed in require at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI.pm line 21.
BEGIN failed--compilation aborted at S:/Open_Source_Software/Slic3r/Slic3r-Prusa3D_1.31.2/lib/Slic3r/GUI.pm line 21.
Compilation failed in require at (eval 103) line 1.

0.6703:
Gives over 10000 "Unable to close this loop having (1..4) points" messages, and looks like this

[image]

but should look like this:

[image]

0.6704:
Behaves the same as 0.6703.

0.70:
Caused the seg fault described yesterday.

lordofhyphens, 2016-11-14:
Take it over to the prusa3d issue tracker unless you can reproduce it with
our master branch, not theirs. They have done quite a bit recently to it.

bubnikv, 2016-11-14:
I just verified with the Prusa3D 1.31.5 32bit Windows build, which has been built with Perl OpenGL 0.6704 and I don't see such a problem. No note of bad triangles. I believe you downloaded the model from here:
http://www.thingiverse.com/thing:922740/#files

It may very well be, that it is a 64bit specific problem. @lordofhyphens is right, if you are mentioning the Prusa3D fork, you should create an issue at the Prusa3D fork, please, even though we may find out that the alexrj master does the same thing in 64 bit builds.

lordofhyphens, 2016-11-15:
Yeah, if we can reproduce it with a 64-bit build of alexrj/master, then by
all means it goes here :).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

bubnikv, 2016-11-15:
> 0.6703: Gives over 10000 "Unable to close this loop having (1..4) points" messages, and looks like this

I have added an issue into Prusa3D issue tracker.
https://github.com/prusa3d/Slic3r/issues/33
Please add to this issue if you find any similar case where the Prusa3D build gives more errors than the alexrj build. It may well be compiler related (different rounding of FP calculations) and it may be 64bit dependent (admesh not 64bit safe).

bubnikv, 2016-11-18:
@bwbeall

> 0.6703:
>  Gives over 10000 "Unable to close this loop having (1..4) points"
messages, and looks like this

This was a Visual Studio 2013 C++ compiler issue. I did not have this issue
with debug builds, but with optimization enabled, the admesh STL import was
broken. I am not quite sure whether the code was fishy or the compiler.
Well, the admesh code was certainly fishy at multiple spots, some of the
code would run on big endian machines, some not, some would not be safe in
regard to floating point compares, the mesh rotation function was
unorthodox, comparing floats using memcmp does not compare +0 and -0 as
equals etc. I tried to fix some of the known and potential issues in the
Prusa3D fork.

The Prusa3D fork certainly shows more "erors fixed" when loading STLs, but
I verified that the errors are real. The error statistics must get lost in
the alexrj Slic3r. Also the Prusa3D shall now connect the triangles into
patches more reliably due to my admesh patches, so the latest Prusa3D code
shows a bit less fixed mesh errors then before.

bwbeall, 2016-11-21:
Hi bubnikv and lordofhyphens,

I have successfully built and loaded the prusa3d slic3r! I have put my instructions in prusa3d/slic3r issue #29. Thanks for all of your assistance getting this going. Please let me know if you have any questions or if there's anything I can do to help.

lordofhyphens, 2016-11-21:
Glad to hear that. Closing :)

closed 2016-11-21 by lordofhyphens