slic3r/Slic3r #3559: Feature Request: Implement Varislice

mrjohnc, 2016-11-04:
Implement Varislice as described at http://hackaday.com/2016/11/03/variable-thickness-slicing-for-3d-printers/

Patola, 2016-11-04:
@platsch is working on it in #1386

lordofhyphens, 2016-11-04:
@platsch was working on it before it was cool.

Pretty sure the second part is NP hard.

platsch, 2016-11-04:
@lordofhyphens maybe this is a misunderstanding, but what makes you think dynamic infill would be computationally hard? Actually, I think the current implementation already works more or less this way. You can just set `infill_every_n_layers` to a high number (e.g. 10) and the algorithm will combine as many variable perimeter layers as possible to a single infill layer.
Works pretty well in my experiments.

lordofhyphens, 2016-11-04:
I am probably thinking of a different problem definition, but I thought
that the request was to determine the maximum number of layers to combine.

lordofhyphens, 2016-11-08:
Platsch's work in #1386 does this better, and setting combine infill and setting it to a high number maximixes infill.

closed 2016-11-08 by lordofhyphens

mrjohnc, 2016-11-08:
Thanks @lordofhyphens