slic3r/Slic3r #3560: 3D Lab Print plane compatible

AndWik, 2016-11-05:
### Version
_Version of Slic3r used goes here_
1.2.9

_Use `About->About Slic3r` for release versions_
1.2.9

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 64-bit

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
3DLabPrint planes are not compatible.
slic3r is really the best 3D print program there is so please make support for this.

Thanks.
Anders
Sweden

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2016-11-06:
Nowhere near enough information to begin, see contributing guidelines. Need, at minimum, a screenshot of the resulting toolpaths (if not the model itself as you wouldn't have the rights to do so).

If I don't see any other activity in a week this is getting closed as "not enough information to fix".

bubnikv, 2016-11-06:
I know the 3DLabPrint guy in person, he visited us at our Prague office and
he received the I3 MK2 printer to fine tune his models.

He designs 3D printed flying RC airplane modes. He is also an architect and
a hobby pilot.
https://3dlabprint.com/

His models are to be printed with a single perimete only for the wing skin
and with a single perimeter for each of the internal strucures (ribs). The
STLs he generates are not quite valid solids. They contain a solid for the
wing enclosing all the ribs, and then all the ribs as separate solids. He
fine tuned the gaps between the wing and the ribs, so the Simplify3D, Cura
or the MatterSlicer do not merge them into a single solid, but they slice
them separately instead. Slic3r on the other hand removes the internal
structures.

In my opinion it makes sense to borrow the mesh processing code from Cura
and make it at least switchable. For some models the Slic3r "mouth" is
better, for some models the Cura way is better.

Vojtech

AndWik, 2016-11-06:
Thank you very much for your explanation. Now I understand the reason.

Have a good day
Anders.
Sweden

bubnikv, 2016-11-06:
I think this issue may stay open as it is valid.

lordofhyphens, 2016-11-06:
Agreed, if someone with a test case is willing to work on it. Probably would be good to add an auto test as well.

epatel, 2016-11-06:
@AndWik Tried this? https://tools3d.azurewebsites.net it has worked very well when others have failed...

AndWik, 2016-11-22:
Here is a link to a Wing Tests made by 3D Lab Print that is told not to work with Slic3r,
http://www.thingiverse.com/thing:998525
http://www.thingiverse.com/thing:1286190

This is a rework that is made for Slic3r
http://www.thingiverse.com/thing:1207209

Thanks
Anders
Sweden

alexrj, 2016-11-22:
Well, this issue has a bad title and poor information. And it looks like @AndWik didn't even test personally.

@bubnikv, my basic opinion is that bad solids should be fixed at CAD level and people should just learn proper usage of CAD programs. But I understand that end-users bring a variety of test cases, especially when they are customers and not advanced users.  Moreover, I have to admit I eased workarounds and tricks also in other features of Slic3r so I do see the value in allowing shortcuts.

So let's try to figure out what a proper feature request would be. What about "Alternative handling of nested solids or non-solid walls"? There could be an option or an interactive prompt when such objects are imported in the plater. This needs a bunch of significant test cases, and sound proposals.   It's not just a matter of mesh processing code; it would significantly affect the rest of the internals as much of the existing semantics would need to be adjusted (i.e. what's an external perimeter? what's a bottom solid layer? what to draw first? etc.)

JBFUK, 2016-12-14:
In addition from what I have seen 3DLabPrint specify that the first few mm should be printed with 2 perimeters with subsequent layers changing to 1 perimeter.  I have seen this sort of functionality being used in both cura and Simplify3D for various purposes like this example, or, perhaps changing temperatures or extrusion multiplier at certain Z heights.  It's a real shame Slic3r can't do this currently.  As quite a newbie to 3D printing I have just managed to get things working well with Slic3r so don't really have the appetite for starting to learn/tune a new slicer.

lordofhyphens, 2016-12-14:
Modifier meshes. I have a new PR to make using them easier.

JBFUK, 2016-12-14:
Great, I have just read some info on this.  Very interesting but not as straightforward as the method of specifying different settings for different Z ranges in Simplify3D.  Still I'm going to experiment and see how I can use this.  Looking at your PR I'm not sure, would it allow me to specify lets say a cube covering the full print area form lets say z=5 to z=100 and specify modifiers for that area?  If so that sounds great.

As per the original poster I'm interested in the 3D Lab Print models but I'm in no hurry, particularly if at some point in the not too distant future I would be able to print them without moving to another slider.

lordofhyphens, 2016-12-14:
Yeah, add a slab modifier for height of 95 and then shift it up to z=5. Add
your overrides there.

You can do it right now without my patch but you would need to open up CAD
sw to create the mesh. My patch just lets you skip that step and do it in
Slic3r.

It's a lot more powerful than the z range solution implemented by s3d.

Could probably ape that solution though with an adaptation of my new setup
if you so desired.

lordofhyphens, 2017-01-25:
Looks like #3523 is this in actuality (hackish STLs, etc). Tagging and closing. Further discussion can go there.

closed 2017-01-25 by lordofhyphens