slic3r/Slic3r #3562: UI is cropped/bad visualized when Windows character/font zoom is active

dfiori, 2016-11-07:
### Version
1.2.8

### Operating system type + version
Windows 7 x64

### Behavior
Some parts of the UI are cropped/not visible due to the incorrect support of Windows font/charatcters zooming feature (actually I set it to 125%).

lordofhyphens, 2016-11-07:
I can haz screenshot showing problem?

noisemaker00, 2016-11-12:
Hi guys, same problem here (slic3r 1.2.9).
I made a screenshot to show the issue.
[image]

bubnikv, 2017-02-09:
I actually cannot reproduce the issue on Windows 10, but I have seem the same issue on Windows 7 of my boss.

noisemaker00, 2017-02-09:
Yep, I have Win 7 x64

bhurler1, 2017-12-21:
I am also having this same issue on **Windows 10 x64**.

To reproduce:

1. Right click on the desktop and click "Display Settings".
2. In the window that opened, look for the heading "Scale and Layout".
3. Select the option, "custom scaling".
4. In the box provided, type 125, and click Apply.
5. Sign out of the Windows profile and sign back in or restart your PC.
6. Open Prusa Slic3r edition.
7. The box should now be displayed incorrectly, shown below.

[image]

This happens in both the standard edition & multi-material.

Zemistr, 2019-06-20:
Any progress? It's really bad with 4k screen and 150 % zoom :(