slic3r/Slic3r #3563: Add optimal layer height for your Z axis (using axis bar pitch/steps/thread starts count)

dfiori, 2016-11-07:
### Version
1.2.9

### Operating system type + version
Windows 7 x64

### Behavior
NEW FEATURE:
let user insert it's Z-Axis bar pitch and thread starts count. When user select a layer height just suggest the layer height related to Z-Axis bar setting closest to the user value.
Something similar to [http://prusaprinters.org/calculator/](url) (@Optimal layer height for your Z axis section)

lordofhyphens, 2016-11-08:
I personally think that this is mostly UI bloat myself.

bubnikv, 2016-11-08:
Some machines are sensitive to it as the micro stepping may not be precise enough. The Prusa I3 runs a continous Z bed leveling and this does not seem to be a problem. So YMMV.

lordofhyphens, 2016-11-18:
Just saying that I'm not particularly keen on adding this to the UI; it's got a relatively narrow use case and a user should know/learn their printer well enough to just use the calculator.

Patola, 2016-11-18:
I also think it is UI bloat but I do not think it is a bad request. It might be of use if/when slic3r merges @platsch 's adaptive slicing algorithm. Maybe it can serve as a template for the calculation of intermediate layers? Maybe some option like "use optimal layer of [___] mm" or "Try to maintain optimal layer sizes for [__] pitch [__] threads [__] microstepping"?

noisemaker00, 2016-11-18:
I read more than one topics about the Z-Layer height, and very often users suggest to keep it a multiple of motor FULL step. You can read something here

- https://www.reddit.com/r/3Dprinting/comments/3zqwii/til_make_sure_your_layer_height_is_a_multiple_of/

- https://www.evernote.com/shard/s211/sh/701c36c4-ddd5-4669-a482-953d8924c71d/1ef992988295487c98c268dcdd2d687e

I think it's not a bad idea adding this suggestion near the layer height selection. Maybe you can set an 'expert mode' to show it only in that mode.

bubnikv, 2016-11-18:
>    til_make_sure_your_layer_height_is_a_multiple_of/
>    4669-a482-953d8924c71d/1ef992988295487c98c268dcdd2d687e

This sounds like a machine configuration to me.

alexrj, 2017-03-17:
We now have a z_steps_per_mm options which does what you want. Thank you for the idea.

closed 2017-03-17 by alexrj