slic3r/Slic3r #3564: Segmentation fault on start

jschall, 2016-11-08:
### Version
Master, stable, 1.2.9 from Ubuntu repo

_Use `About->About Slic3r` for release versions_
Cannot

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_
1.2.9-34-g49f327f
1.2.9-367-g8910b49

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 16.04

### Behavior
* _Describe the problem_
Ubuntu package was working, decided to try master, followed the build instructions, and now neither version will start. I've tried removing ~/.Slic3r - when I do that, a window briefly flashes on the screen but the result is the same.

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_

* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

jschall, 2016-11-08:
Also tried:

    sudo apt-get autoremove --purge slic3r 
    sudo apt-get install slic3r

Did not help. At this point I'm happy to be up and running with any version of slic3r.

lordofhyphens, 2016-11-08:
Don't know what to tell you re: the Ubuntu repo version; you should also
report this to that maintainer.

Consider the linux download from slic3r.org or building 1.3.0-dev from
source.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

jschall, 2016-11-08:
Isn't master 1.3.0-dev? I did try that. They're all segfaulting. I did just do the linux download from slic3r.org - it works, so at least I have *something* that works.

lordofhyphens, 2016-11-08:
See #3540

closed 2016-11-08 by lordofhyphens