slic3r/Slic3r #3570: Bridge slicing path messed up

sudoeste, 2016-11-10:
### Version
Latest from repetier-host, it says latest when trying to update inside slic3r settings

### Operating system type + version
Windows 10

### Behavior
Slice the following model
http://www.thingiverse.com/thing:1729459
* _Expected Results_
the first bridge layer should be uniform (Cura does it)
* _Actual Results_
[image]
_Is this a new feature request?_
no

lordofhyphens, 2016-11-10:
Need a screenshot from inside Slic3r itself.

sudoeste, 2016-11-10:
Sure thing. All default settings
[image]

lordofhyphens, 2016-11-10:
Have you tried with a 1.3.0-dev version? 
https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/1.3.0-master-183#files
[image]

lordofhyphens, 2016-11-10:
Just FYI, the 2d layer view won't  show you previous layers, but 3D will (3D will also show you what Slic3r expects the widths to be). 
Configuration: [config.zip](https://github.com/alexrj/Slic3r/files/582482/config.zip)

sudoeste, 2016-11-10:
I'm new to this world :) 

Nop, I haven't tried the dev build, just the download from the main site. So I guess it's fixed but only in the 1.3 branch?

lordofhyphens, 2016-11-10:
Apparently so.

This is roughly equivalent to building the source yourself (there are
instructions for doing so).

It should be possible to use the dev version of slic3r with RepHost (point
it at the batch file instead of an exe). I haven't tried though. I don't
have any use for RepHost myself.

sudoeste, 2016-11-10:
I'm new to the 3d world not how building from source works (although I can't program anything even if my life depends on it) . 

So this probably won't be fixed upstream?

lordofhyphens, 2016-11-10:
You are upstream :)

It should be fixed in the next release, whenever that is. If you want to
get the benefit before that, there is the bintray.com builds (which I
linked earlier).

Take a look at the wiki tab here on github, there are instructions for
running from git on windows, which is what you would need to do if you want
to build it yourself. If the bintray.com builds serve your needs, then
great :)

micscopau, 2016-11-12:
I'm having a similar issue that I would describe as a bad bridge slicing path 

**Version:** slic3r 1.3.0-dev 

trying to print:
http://www.thingiverse.com/thing:1179160

Internally there is a solid perimeter that bridging could use for anchoring (first picture), but instead has bridge paths stopping above free space (second picture). 

Thanks!

[image]

[image]

bubnikv, 2016-11-12:
This issue has been fixed to a large extent in the recent prusa3d fork. If
the bridging regions are very close, then they are merged together.

[image: Inline image 1]

bubnikv, 2016-11-21:
Duplicate of https://github.com/alexrj/Slic3r/issues/3578

lordofhyphens, 2017-01-25:
Going to close as #3578 has a longer comment trail and I can't reproduce with the original file on 1.3.0-dev.

closed 2017-01-25 by lordofhyphens