slic3r/Slic3r #3573: IO::Stringy and IO::Scalar on windows won't compile whatever version of boost & perl I use

arhi, 2016-11-13:
### Version
git (same for master and stable)

e:\bin\slic3r-dev>git rev-parse HEAD
49f327f0e0c608401060a33d5bffa438bc1985bd

### Operating system type + version
WIN7 64bit

### Behavior
IO::Scalar - not compiling

a. tried citrusperl-standard-51603 (nothing older available)
b. tried strawberry-perl-5.18.4.1-64bit
c. tried strawberry-perl-5.18.0.1-64bit
d. boost is boost_1_59_0

1. install perl (a/b/c)
2. install boost to c:\dev\boost_1_59_0
3. get clone slic3r
4. perl Build.pl
... (bunch of OK's)
--> Working on ./xs
Entering E:/bin/slic3r-dev/xs
Configuring E:/bin/slic3r-dev/xs ... Running Build.PL
'cc' is not recognized as an internal or external command,
operable program or batch file.
Checking prerequisites...
  **build_requires:
    !  ExtUtils::Typemaps::Basic is not installed
    !  ExtUtils::Typemaps::Default is not installed**

ERRORS/WARNINGS FOUND IN PREREQUISITES.  You may wish to install the versions
of the modules indicated above before proceeding with this installation
...
...
Files=19, Tests=456,  2 wallclock secs ( 0.11 usr +  0.01 sys =  0.12 CPU)
Result: PASS
Building Slic3r-XS
Files found in blib\arch: installing files in blib\lib into architecture dependent library tree
Installing E:\bin\strawberry\perl\site\lib\auto\Slic3r\XS\XS.bs
Installing E:\bin\strawberry\perl\site\lib\auto\Slic3r\XS\XS.dll
Installing E:\bin\strawberry\perl\site\lib\Slic3r\XS.pm
OK
Successfully installed Slic3r-XS-0.01
2 distributions installed
The following prerequisites failed to **install: IO::Scalar**

I try then to install IO::Scalar manually:
e:\bin\slic3r-dev>cpan

cpan shell -- CPAN exploration and modules installation (v2.00)
Enter 'h' for help.

cpan> install IO::Scalar
Fetching with LWP:
http://cpan.strawberryperl.com/authors/01mailrc.txt.gz
Fetching with LWP:
http://cpan.strawberryperl.com/modules/02packages.details.txt.gz
Fetching with LWP:
http://cpan.strawberryperl.com/modules/03modlist.data.gz
Creating database file ...
Done!Running install for module 'IO::Scalar'
Running make for D/DS/DSKOLL/IO-stringy-2.111.tar.gz
Fetching with LWP:
http://cpan.strawberryperl.com/authors/id/D/DS/DSKOLL/IO-stringy-2.111.tar.gz
Fetching with LWP:
http://cpan.strawberryperl.com/authors/id/D/DS/DSKOLL/CHECKSUMS
Checksum for E:\bin\strawberry\cpan\sources\authors\id\D\DS\DSKOLL\IO-stringy-2.111.tar.gz ok
Scanning cache E:\bin\strawberry\cpan\build for sizes
DONE

  CPAN.pm: Building D/DS/DSKOLL/IO-stringy-2.111.tar.gz

Checking if your kit is complete...
Looks good
Generating a dmake-style Makefile
Writing Makefile for IO::Stringy
Writing MYMETA.yml and MYMETA.json
cp lib/IO/WrapTie.pm blib\lib\IO\WrapTie.pm
cp lib/IO/Stringy.pm blib\lib\IO\Stringy.pm
cp lib/IO/ScalarArray.pm blib\lib\IO\ScalarArray.pm
cp lib/IO/AtomicFile.pm blib\lib\IO\AtomicFile.pm
cp lib/IO/Scalar.pm blib\lib\IO\Scalar.pm
cp lib/IO/Lines.pm blib\lib\IO\Lines.pm
cp lib/IO/InnerFile.pm blib\lib\IO\InnerFile.pm
cp lib/IO/Wrap.pm blib\lib\IO\Wrap.pm
  DSKOLL/IO-stringy-2.111.tar.gz
  E:\bin\strawberry\c\bin\dmake.exe -- OK
Running make test
"E:\bin\strawberry\perl\bin\perl.exe" "-MExtUtils::Command::MM" "-MTest::Harness" "-e" "undef *Test::Harness::Switches; test_harness(0, 'blib\lib', 'b
lib\arch')" t/*.t
t/IO_InnerFile.t .... Failed 7/7 subtests
t/IO_Lines.t ........ ok
t/IO_Scalar.t ....... ok
t/IO_ScalarArray.t .. ok
t/IO_WrapTie.t ...... ok
t/simple.t .......... ok
t/two.t ............. ok

Test Summary Report
-------------------
t/IO_InnerFile.t  (Wstat: 0 Tests: 7 Failed: 7)
  Failed tests:  1-7
Files=7, Tests=125,  1 wallclock secs ( 0.03 usr +  0.00 sys =  0.03 CPU)
Result: FAIL
Failed 1/7 test programs. 7/125 subtests failed.
dmake.exe:  Error code 255, while making 'test_dynamic'
  DSKOLL/IO-stringy-2.111.tar.gz
  E:\bin\strawberry\c\bin\dmake.exe test -- NOT OK
//hint// to see the cpan-testers results for installing this module, try:
  reports DSKOLL/IO-stringy-2.111.tar.gz
Running make install
  make test had returned bad status, won't install without force
Stopping: 'install' failed for 'IO::Scalar'.
Failed during this command:
 DSKOLL/IO-stringy-2.111.tar.gz               : make_test NO

cpan> quit

of course it fails :(

lordofhyphens, 2016-11-13:
Not a Slic3r problem per se, but I will take a look at the build log.

lordofhyphens, 2016-11-13:
Something you can do is copy gcc.exe to cc.exe in the mingw dir of
strawberry perl.
On Nov 13, 2016 1:05 AM, "Joe Lenox" <lordofhyphens@gmail.com> wrote:

> Not a Slic3r problem per se, but I will take a look at the build log.
>

arhi, 2016-11-13:
hi mate, the part I think is "part of slic3r" is the doc (how to build on windows), some versions of prerequisites are no longer available for download so maybe that page should be updated with "what up to date / available versions of perl, boost etc.. win build will work with" ... I'm attm compiling with 32bit latest strawberry, at least the IO::Scalar is there :D

arhi, 2016-11-13:
update ..

with 32bit (latest) stawberry it finish building but crashes immediately on start

e:\bin\slic3r-dev>perl Build.PL
App::cpanminus is up to date. (1.7042)
Class::XSAccessor is up to date. (1.19)
Encode is up to date. (2.87)
Encode::Locale is up to date. (1.05)
ExtUtils::MakeMaker is up to date. (7.24)
ExtUtils::ParseXS is up to date. (3.30)
File::Basename is up to date. (2.85)
File::Spec is up to date. (3.63)
Getopt::Long is up to date. (2.49)
IO::Scalar is up to date. (2.111)
Math::PlanePath is up to date. (123)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.002005)
POSIX is up to date. (1.65)
Scalar::Util is up to date. (1.46)
Test::Harness is up to date. (3.36)
Test::More is up to date. (1.302062)
Thread::Semaphore is up to date. (2.13)
Time::HiRes is up to date. (1.9740)
Unicode::Normalize is up to date. (1.25)
Could not find the MinGW directory at C:\dev\CitrusPerl\mingw32; skipping XML::SAX::ExpatXS (only needed for faster parsing of AMF files)
XML::SAX::ExpatXS is up to date. (1.33)
threads is up to date. (2.09)
Warning: Build.PL has been altered.  You may need to run 'perl Build.PL' again.
Cleaning up build files
Cleaning up configuration files
cpanm (App::cpanminus) 1.7042 on perl 5.024000 built for MSWin32-x86-multi-thread-64int
Work directory is e:\Users\arhimed/.cpanm/work/1479023893.37600
You have make E:\bin\Strawberry\c\bin\dmake.exe
You have LWP 6.15
Falling back to Archive::Tar 2.06
--> Working on ./xs
Entering E:/bin/slic3r-dev/xs
Checking if you have ExtUtils::Install 1.46 ... Yes (2.04)
Checking if you have Module::Build 0.38 ... Yes (0.4218)
Running Build.PL
Configuring E:/bin/slic3r-dev/xs ... 'cc' is not recognized as an internal or external command,
operable program or batch file.
Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.30)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.30)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have Module::Build 0.3601 ... Yes (0.4218)
Checking if you have Test::More 0 ... Yes (1.302062)
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
...
Generating script 'buildtmp\XS.lds'
dlltool --def "buildtmp\XS.def" --output-exp "buildtmp\XS.exp"
g++ -o "blib\arch\auto\Slic3r\XS\XS.xs.dll" -Wl,--base-file,"buildtmp\XS.base" -Wl,--image-base,0x2b7d0000 -mdll -s -L"E:\bin\STRAWB~1\perl\lib\CORE"
-L"E:\bin\STRAWB~1\c\lib" "buildtmp\XS.lds" -lstdc++ "buildtmp\XS.exp"
dlltool --def "buildtmp\XS.def" --output-exp "buildtmp\XS.exp" --base-file "buildtmp\XS.base"
g++ -o "blib\arch\auto\Slic3r\XS\XS.xs.dll" -Wl,--image-base,0x2b7d0000 -mdll -s -L"E:\bin\STRAWB~1\perl\lib\CORE" -L"E:\bin\STRAWB~1\c\lib" "buildtmp
\XS.lds" -lstdc++ "buildtmp\XS.exp"
t\01_trianglemesh.t ............. ok
t\03_point.t .................... ok
t\04_expolygon.t ................ ok
t\05_surface.t .................. ok
t\06_polygon.t .................. ok
[... 94 lines cut ...]
Configuring Socket-2.024 ... OK
Building and testing Socket-2.024 ... OK
Successfully installed Socket-2.024 (upgraded from 2.021)
1 distribution installed
Win32::API is up to date. (0.84)
--> Working on Wx
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Wx-0.9928.tar.gz ... OK
==> Found dependencies: Alien::wxWidgets
--> Working on Alien::wxWidgets
Fetching http://www.cpan.org/authors/id/M/MD/MDOOTSON/Alien-wxWidgets-0.67.tar.gz ... OK
Configuring Alien-wxWidgets-0.67 ... OK
Building and testing Alien-wxWidgets-0.67 ... OK
Successfully installed Alien-wxWidgets-0.67
Configuring Wx-0.9928 ... OK
Building and testing Wx-0.9928 ... OK
Successfully installed Wx-0.9928
2 distributions installed
--> Working on Wx::GLCanvas
Fetching http://www.cpan.org/authors/id/M/MB/MBARBON/Wx-GLCanvas-0.09.tar.gz ... OK
Configuring Wx-GLCanvas-0.09 ... OK
Building and testing Wx-GLCanvas-0.09 ... OK

1 distribution installed

e:\bin\slic3r-dev>perl slic3r.pl

lordofhyphens, 2016-11-13:
> 96.tar.gz ... OK
> wxWidgets-0.67.tar.gz ... OK
> 09.tar.gz ... OK

The crash on start is likely due to #3540. Apply the changes from
https://github.com/alexrj/Slic3r/pull/3552/files to Build.PL. You may need
to uninstall OpenGL.pm (see the comment thread).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-11-13:
As for the test failures on IO:Scalar/IO::Stringy, please report your issue to the maintainer of those packages:
http://search.cpan.org/~dskoll/IO-stringy-2.111/lib/IO/Stringy.pm

This isn't something we fix on our end. 

http://cpantesters.org/distro/I/IO-stringy.html Looks like this failure may be known. You are welcome to try to force installation.

arhi, 2016-11-13:
@lordofhyphens hi mate, sorry if my report came out wrong :D I was more in the "versions you suggest are either unavailable or don't work" for more like fixing the documentation about win-build, not about fixing the stringy .. and stringy (and scalar) work properly on recent releases so I doubt maintainer cares much about version from 5 years ago not working properly today :( .. 

anyhow, the initial crash is 100% because of the opengl 0.7 having some incompatible changes so I'm trying to get this to clean, recompile now with older .. I'm not very windows+mingw savy so everything takes time :( .. linux is so easier to develop on .. but I need to setup this on windows now so.. 

thanks for the support :)

arhi, 2016-11-13:
ah and it does work now with older opengl so as for the "won't start" that's definitely #3540 .. as for the "how to build on windows in end of 2016" ..

alexrj, 2016-11-22:
The fix for #3540 is now merged in HEAD.

The IO::Scalar installation issue should be reported to its maintainer. I'm sorry but we can't do much here... I hope you don't mind if I close this issue report.  :)

As @lordofhyphens suggested you might try to force the installation of IO::Scalar regardless of the tests. To do so, type `cpanm -f IO::Scalar` or `cpan` followed by `force install IO::Scalar` (two different methods).

closed 2016-11-22 by alexrj

arhi, 2016-11-22:
I solved the problem by using latest Strawberry Perl 5.24.0.1-32bit and with it it works, with 64bit or the version suggested on the wiki for windows setup does not :( .. so yes close def. not slic3r problem .. apart from maybe updating the doc on windows compiling from source .. Strawberry Perl 5.24.0.1-32bit works properly on both 64bit and 32bit windows :D