slic3r/Slic3r #3577: [filament_preset] not multi-valued with multiple extruders

hexane360, 2016-11-19:
### Version
1.2.9
### Operating system type + version
Windows 10 Home Insider Preview Build 14931
### Behavior

I would like to create diagnostic start g-code that displays what filaments I am printing with. However, [filament_preset] currently only returns the filament profile for extruder 0, and cannot be indexed. Is there a workaround for this?

#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/alexrj/Slic3r/files/601319/config.zip)

bubnikv, 2016-11-19:
Here is the description of macro expansions available in Slic3r
https://github.com/alexrj/Slic3r/wiki/FAQ#what-placeholders-can-i-use-in-custom-g-code
http://mauk.cc/mediawiki/index.php/Slic3r_placeholders

You are right that it is currently not possible to expand multiple filament
preset file names.Following placeholders are expanded per filament though:

filament_colour: a hexadecimal RGB value

filament_diameter: this is obvious

filament_settings_id: set to empty string, not used for anything useful.

So the goal here obviously is to make use of the unfinished
filament_settings_id. My proposal is to bring it to the UI. If the value is
undefined, initialize it with the preset name. Otherwise leave it intact.

Vojtech

bubnikv, 2016-11-19:
By the way, the Slic3r Prusa fork has a new filament_notes field, which you can use for your purpose right away.

alexrj, 2017-03-26:
Done. [filament_preset_0], [filament_preset_1] .. [filament_preset_n] are now populated as well as [filament_preset] (which holds the first one).

closed 2017-03-26 by alexrj