slic3r/Slic3r #3578: Wrong multiple bridges merging.

diggit, 2016-11-20:
### Version
1.3.0-dev built from master branch

### Operating system type + version
Archlinux, up to date.

### Behavior
* **Problem:** Intersecting bridging areas produces path turns above gaps. (by bridging area is meat area containing all bridge paths above one hole)
* _Steps needed to reproduce the problem_
  * Slice model with holes to bridge close to each other. (Used text in screenshot.)
* _Expected Results_
  * Continuous bridges across whole holes.
* _Actual Results_
[image]

IMO, solution would be processing all intersecting bridging areas as one and not separately

[slic3r_bridge_err.zip](https://github.com/alexrj/Slic3r/files/601814/slic3r_bridge_err.zip)

bubnikv, 2016-11-20:
> IMO, solution would be processing all intersecting bridging areas as one
and not separately

I have fixed that in the Prusa3D fork exactly the way you proposed it,
although the implementation was more involved.

Richie76, 2016-11-20:
Hi,

I just came across a similar issue using the Prusa3D Fork.
Exact version: 1.30.0.25-prusa3d-release_candidate_1_3-win64

Please see the attached picture for reference:
[wrong layer]http://www.loopback.at/wp-content/uploads/2016/11/layer_problem_slic3er-1024x752.jpg)

this is the underside of a lego duplo brick - i did some 4x4 too - there slic3r tried to print into the holes first.

could you please look into this ?

rgds, richard

bubnikv, 2016-11-20:
> I just came across a similar issue using the Prusa3D Fork.
> Exact version: 1.30.0.25-prusa3d-release_candidate_1_3-win64

That is an old build. The latest official build is 1.31.5
https://github.com/prusa3d/Slic3r/releases/tag/version_1.31.5

Richie76, 2016-11-20:
ok ill download it and test it. thanks for the quick relpy

diggit, 2016-11-20:
@bubnikv cool. I am sure it was not so simple. When you solved it, why it isn't in upstream yet?

lordofhyphens, 2016-11-21:
@diggit Hi, I am the person working on that.

The changes are generally not easy to separate from everything else he's
done so there is a pile of work on me to get them packaged upstream.

Then they need to be reviewed and approved by @alexrj. We are both people
working in our spare time on this, @bubnikv has it as his job.

bubnikv, 2016-11-21:
Duplicate of https://github.com/alexrj/Slic3r/issues/3570

closed 2018-08-26 by diggit